TrunkRecorder/trunk-recorder #665: Making UHD, USRP requirements optional

rosecitytransit, 2022-04-11:
Attached is a patch that removes the UHD/USRP code sections from the current master branch in order to be able to compile without having support for those more expensive SDRs.

If there's a desire and it's easily to do, I was wondering if those sections could be made optional.

[0001-compiles-w-o-UHD-USRP-support.patch.txt](https://github.com/robotastic/trunk-recorder/files/8460508/0001-compiles-w-o-UHD-USRP-support.patch.txt)

robotastic, 2022-04-30:
It is pretty hard for me to track the changes with a Patch - can you submit it as PR?

The UHD/USRP is being used so, it would need to be included by default. There could be something with a CMAKE flag to optional disable it.... but so far I haven't done CMAKE flags and it creates additional scenarios to test against. 

Can you explain more about why you are interested in this?

rosecitytransit, 2022-04-30:
I haven't actually investigated how to set things up to do it.  I definitely agree that by default UHD/USRP should be supported; I was thinking of something like "if (UHD libraries exist) then include UHD code".

Since I had no plans to use anything besides RTL devices, I installed GNU Radio without support for other types (using https://github.com/robotastic/trunk-recorder/wiki/MinimalInstall ), and as a result trunk-recorder will not compile unless I apply the patch.

closed 2024-12-14 by robotastic