TrunkRecorder/trunk-recorder #669: StatusServer Segmentation Fault - Mac M1

tadscottsmith, 2022-04-14:
On a Mac M1 (ARM-64) as soon as I enter a statusServer, trunk-recorder crashes. It appears to be something with the poll_one() function, but I'm a little out of my league trying to debug this. I never see the ``` [Connect] Successful Connection``` in the logs. The statusServer is on a remote computer, and I'm able to use the same statusServer with the same config file to successfully run trunk-recorder on Ubuntu.

```
PLUGINS
-------------------------------------
[2022-04-14 18:45:42.151464] (info)   Setting up plugin -  Name: openmhz_uploader	 Library file: libopenmhz_uploader.so
[2022-04-14 18:45:42.152167] (info)   Setting up plugin -  Name: broadcastify_uploader	 Library file: libbroadcastify_uploader.so
[2022-04-14 18:45:42.152728] (info)   Setting up plugin -  Name: unit_script	 Library file: libunit_script.so
[2022-04-14 18:45:42.153734] (info)   Setting up plugin -  Name: stat_socket	 Library file: libstat_socket.so
[2022-04-14 18:45:42.163636] (info)   No plugins configured
[2022-04-14 18:45:42.163661] (info)   

[2022-04-14 18:45:42.163754] (info)   [ramsey]	Started with Control Channel: 8.520875e+08
[2022-04-14 18:45:42.164109] (debug)   P25 trunking Decim: 25 Decim2:  4
[2022-04-14 18:45:42.164117] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 96000 Second decimated rate: 24000 FA: 6250 FB: 12000 System Rate: 2400000
[2022-04-14 18:45:42.167284] (info)   	 P25 Trunking ARB - Initial Rate: 2400000 Resampled Rate: 24000 Initial Decimation: 25 System Rate: 24000 ARB Rate: 1
[2022-04-14 18:45:42.168306] (info)   [mplscc]	Started with Control Channel: 8.602375e+08
[2022-04-14 18:45:42.168643] (debug)   P25 trunking Decim: 25 Decim2:  4
[2022-04-14 18:45:42.168653] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 96000 Second decimated rate: 24000 FA: 6250 FB: 12000 System Rate: 2400000
[2022-04-14 18:45:42.171926] (info)   	 P25 Trunking ARB - Initial Rate: 2400000 Resampled Rate: 24000 Initial Decimation: 25 System Rate: 24000 ARB Rate: 1
zsh: segmentation fault  ./trunk-recorder
```

```-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               trunk-recorder [70580]
Path:                  /Users/USER/*/trunk-recorder
Identifier:            trunk-recorder
Version:               ???
Code Type:             ARM-64 (Native)
Parent Process:        zsh [68862]
Responsible:           Terminal [67604]
User ID:               502

Date/Time:             2022-04-14 18:23:36.1658 -0500
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Anonymous UUID:        EA21F757-E930-F732-FF84-81C7BAA907E5

Sleep/Wake UUID:       AFE67AD6-94C0-423D-97CB-F1E028B47128

Time Awake Since Boot: 75000 seconds
Time Since Wake:       74560 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x000000004d555482
Exception Codes:       0x0000000000000001, 0x000000004d555482
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [70580]

VM Region Info: 0x4d555482 is not in any region.  Bytes before following region: 3076483966
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104b4c000-104d20000    [ 1872K] r-x/r-x SM=COW  ...runk-recorder

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libstat_socket.so             	       0x10f37215c boost::asio::detail::scheduler::post_immediate_completion(boost::asio::detail::scheduler_operation*, bool) + 36 (scheduler.ipp:343)
1   libstat_socket.so             	       0x10f394988 boost::asio::detail::strand_service::do_dispatch(boost::asio::detail::strand_service::strand_impl*&, boost::asio::detail::scheduler_operation*) + 416 (strand_service.ipp:148)
2   libstat_socket.so             	       0x10f394988 boost::asio::detail::strand_service::do_dispatch(boost::asio::detail::strand_service::strand_impl*&, boost::asio::detail::scheduler_operation*) + 416 (strand_service.ipp:148)
3   libstat_socket.so             	       0x10f3f01f4 void boost::asio::detail::strand_service::dispatch<boost::asio::detail::rewrapped_handler<boost::asio::detail::binder2<boost::asio::detail::wrapped_handler<boost::asio::io_context::strand, std::__1::__bind<void (websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>::*)(std::__1::shared_ptr<websocketpp::transport::asio::connection<websocketpp::config::asio_client::transport_config> >, std::__1::shared_ptr<boost::asio::basic_waitable_timer<std::__1::chrono::steady_clock, boost::asio::wait_traits<std::__1::chrono::steady_clock>, boost::asio::any_io_executor> >, std::__1::function<void (std::__1::error_code const&)>, boost::system::error_code const&, boost::asio::ip::basic_resolver_iterator<boost::asio::ip::tcp>), websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>*, std::__1::shared_ptr<websocketpp::transport::asio::connection<websocketpp::config::asio_client::transport_config> >&, std::__1::shared_ptr<boost::asio::basic_waitable_timer<std::__1::chrono::steady_clock, boost::asio::wait_traits<std::__1::chrono::steady_clock>, boost::asio::any_io_executor> >&, std::__1::function<void (std::__1::error_code const&)>&, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&>, boost::asio::detail::is_continuation_if_running>, boost::system::error_code, boost::asio::ip::basic_resolver_results<boost::asio::ip::tcp> >, std::__1::__bind<void (websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>::*)(std::__1::shared_ptr<websocketpp::transport::asio::connection<websocketpp::config::asio_client::transport_config> >, std::__1::shared_ptr<boost::asio::basic_waitable_timer<std::__1::chrono::steady_clock, boost::asio::wait_traits<std::__1::chrono::steady_clock>, boost::asio::any_io_executor> >, std::__1::function<void (std::__1::error_code const&)>, boost::system::error_code const&, boost::asio::ip::basic_resolver_iterator<boost::asio::ip::tcp>), websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>*, std::__1::shared_ptr<websocketpp::transport::asio::connection<websocketpp::config::asio_client::transport_config> >&, std::__1::shared_ptr<boost::asio::basic_waitable_timer<std::__1::chrono::steady_clock, boost::asio::wait_traits<std::__1::chrono::steady_clock>, boost::asio::any_io_executor> >&, std::__1::function<void (std::__1::error_code const&)>&, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&> > >(boost::asio::detail::strand_service::strand_impl*&, boost::asio::detail::rewrapped_handler<boost::asio::detail::binder2<boost::asio::detail::wrapped_handler<boost::asio::io_context::strand, std::__1::__bind<void (websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>::*)(std::__1::shared_ptr<websocketpp::transport::asio::connection<websocketpp::config::asio_client::transport_config> >, std::__1::shared_ptr<boost::asio::basic_waitable_timer<std::__1::chrono::steady_clock, boost::asio::wait_traits<std::__1::chrono::steady_clock>, boost::asio::any_io_executor> >, std::__1::function<void (std::__1::error_code const&)>, boost::system::error_code const&, boost::asio::ip::basic_resolver_iterator<boost::asio::ip::tcp>), websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>*, std::__1::shared_ptr<websocketpp::transport::asio::connection<websocketpp::config::asio_client::transport_config> >&, std::__1::shared_ptr<boost::asio::basic_waitable_timer<std::__1::chrono::steady_clock, boost::asio::wait_traits<std::__1::chrono::steady_clock>, boost::asio::any_io_executor> >&, std::__1::function<void (std::__1::error_code const&)>&, std::__1::placeholders::__ph<1> const&, std::__1::placeholders::__ph<2> const&>, boost::asio::detail::is_continuation_if_running>, boost::system::error_code, boost::asio::ip::basic_resolver_results<boost::asio::ip::tcp> >, std::__1::__bind<void (websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>::*)(std::__1::shared_ptr<websocketpp::transport::asio::connection<websocketpp::config::asio_client [... cut ...]
16  libstat_socket.so             	       0x10f431d50 websocketpp::transport::asio::endpoint<websocketpp::config::asio_client::transport_config>::poll_one() + 28 (endpoint.hpp:664)
17  libstat_socket.so             	       0x10f360fac Stat_Socket::poll_one() + 100 (stat_socket.cc:333)
18  trunk-recorder                	       0x104c9f650 plugman_poll_one() + 128 (plugin_manager.cc:124)
19  trunk-recorder                	       0x104b71a1c monitor_messages() + 1092 (main.cc:1330)
20  trunk-recorder                	       0x104b75b80 main + 5428 (main.cc:1630)
21  dyld                          	       0x105381088 start + 516
```

Dygear, 2022-06-10:
I'll look into this later tonight. Don't have any RTL dongles with me. I'll use Debian arm64 to see if I can produce.

kylegibbons, 2022-10-31:
Any update on this? I am also getting the segmentation fault when trying to connect to a websocket server. The server is on another machine. When I remove the statusServer property from the config, it does not crash. I do not get any other jog messages relating to the seg fault.

tadscottsmith, 2022-10-31:
I was never able to track it down. I see on your Gitter post that you're running into an issue with on an RPI4. What OS version and architecture are you running? The statusServer works on my RPI4 running Raspbian 32bit.

kylegibbons, 2022-10-31:
I am running the latest Raspberry Pi OS List (32-bit)

tadscottsmith, 2022-11-01:
How about your version of Boost? I noticed that my working one has 1.67 and the one that doesn't work is 1.76.
```
cat /usr/include/boost/version.hpp | grep "BOOST_LIB_VERSION"
```

kylegibbons, 2022-11-01:
This is what I get for output:

//  BOOST_LIB_VERSION must be defined to be the same as BOOST_VERSION
#define BOOST_LIB_VERSION "1_74"

kylegibbons, 2022-11-02:
I decided I wanted a beefier machine and upgraded to an x86 machine running Ubuntu Server. I am not having the issue on that platform.

tadscottsmith, 2022-11-02:
What version of Boost is that on the working system?

kylegibbons, 2022-11-02:
//  BOOST_LIB_VERSION must be defined to be the same as BOOST_VERSION
#define BOOST_LIB_VERSION "1_74"

aaknitt, 2022-12-26:
Having the same issue in a Pi.  Segfault as soon as TR starts.  BOOST_LIB_VERSION is 1.74.  

Interestingly, if I use a secure URL such as "https://" or "wss://" instead of "http://" or "ws://" it doesn't crash, but it also doesn't connect to my server because it's not secured.

aaknitt, 2022-12-26:
calling `m_client.poll()` immediately after `m_client.connect(con)` in the plugin's `open(stat)` function does _not_ crash, but calling either `m_client.poll()` or `m_client.poll_one()` inside of the plugin's `poll_one()` function causes an immediate segfault crash.

tadscottsmith, 2022-12-30:
I finally bit the bullet today and uninstalled / reinstalled all the dependencies including gnuradio. I rebuilt and now the statusServer works. Seems like it has to be a dependency version mismatch somewhere, but I'm not even sure where to start.

closed 2024-12-14 by robotastic