TrunkRecorder/trunk-recorder #676: make `nproc` TARGET 4 missing

slinkysavage, 2022-04-26:
When compiling and installing as per the guide...
During this chunk
```
mkdir trunk-build
git clone https://github.com/robotastic/trunk-recorder.git
cd trunk-build
cmake ../trunk-recorder
make -j `nproc`
sudo make install
```

The ``` -j ``` does seize up my Pi 4 B.

When I run it without I get the following error.

``` make:  *** no rule to make target '4'. stop. ```

Any ideas on how to fix this?

taclane, 2022-04-26:
The j option in
```
make -j `nproc`
```
just tells the compiler how many CPUs to use. If your shell seems to be having issues with the make command, just use
```
make -j 4
```
for a raspi4. (Or 3 or 2 or 1 if you want to devote fewer CPUs to compile trunk recorder.)

aaknitt, 2022-04-26:
make -j `nproc` is translated to make -j 4 (nproc = number of processors available = 4 on a Pi 4).  If it's locking up, manually override it to make -j 3 (instead of nproc).  I've also had issues with it locking up when using all 4 processors, so I usually run it with 3.  Even then it'll sometimes error out and when that happens running it again (same command) will succeed.

Dygear, 2022-04-28:
I've never had the issue where it locks up on the Pi 4 (or Pi 3) using all cores. I've always used Samsung Pro Endurance SD Cards.

closed 2022-06-07 by robotastic