TrunkRecorder/trunk-recorder #678: Log for Encrypted calls removed?

kcwebby, 2022-04-29:
Was this comment intentional to remove the log notification for encrypted calls?  I didn't see anywhere listing a reason for change, so I was still expecting to see it in logs.

Thanks!

https://github.com/robotastic/trunk-recorder/blob/dd3e6bfe1c74a7846d6e50c6e98024044d7b64f3/trunk-recorder/main.cc#L627

sally-yachts, 2022-08-10:
I updated one of my recorders to the latest code and I'm also seeing this behavior. I searched discord and it looked like @robotastic mentioned going to fix it on 4/30 but I think it may have gotten lost in some of the no-overlap integration work. 

It looks like we also may have accidentally commented out the "No source covering freq" message as well unless that was relocated. Can reproduce pretty easily by switching between "latest" and "edge" tags of docker image.

robotastic, 2022-09-25:
Should be fixed now.

closed 2022-09-25 by robotastic