TrunkRecorder/trunk-recorder #679: rapidJSON assertion failure

rabarar, 2022-05-02:
Any thoughts on what might be popping this bug?

```trunk-recorder: /usr/include/rapidjson/document.h:1154: rapidjson::GenericValue<Encoding, Allocator>::MemberIterator rapidjson::GenericValue<Encoding, Allocator>::FindMember(const rapidjson::GenericValue<Encoding, SourceAllocator>&) [with SourceAllocator = rapidjson::MemoryPoolAllocator<>; Encoding = rapidjson::UTF8<>; Allocator = rapidjson::MemoryPoolAllocator<>; rapidjson::GenericValue<Encoding, Allocator>::MemberIterator = rapidjson::GenericMemberIterator<false, rapidjson::UTF8<>, rapidjson::MemoryPoolAllocator<> >]: Assertion `IsObject()' failed.
Aborted```

robotastic, 2022-05-30:
This is a weird error - I have never seen Rapid JSON pop up before. Is this something particular to your system?

robotastic, 2022-06-07:
Going to close for now - let me know if this is still an issue

closed 2022-06-07 by robotastic