TrunkRecorder/trunk-recorder #689: SoapySDR SDRPlay Gain Settings not working

W3AXL, 2022-05-17:
FYI I am unable to get gain control to work with my RSP1 from a fresh build of `trunk-recorder`. I've tried several settings recommended in #422 and it appears that the changes in #546 aren't working correctly.

My source config:
```json
    "sources": [{
        "center": 857487500,
        "rate": 8000000,
        "error": 10000,
        "agc": false,
        "gainSettings": {
            "RFGR": 0,
            "IFGR": 25
        },
        "digitalRecorders": 16,
        "driver": "osmosdr",
        "device": "soapy,driver=sdrplay,agc_setpoint=0"
    }],
```

And the output when the device is initialized:
```
[2022-05-16 22:28:12.721095] (info)   Source Device: soapy,driver=sdrplay,agc_setpoint=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file fcd rtl rtl_tcp uhd rfspace soapy redpitaya 
[INFO] devIdx: 0
[INFO] SerNo: B0002P0001
[INFO] hwVer: 1
[INFO] rspDuoMode: 0
[INFO] tuner: 1
[INFO] rspDuoSampleFreq: 0.000000
[INFO] Using format CF32.
[2022-05-16 22:28:12.774237] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2022-05-16 22:28:12.774266] (info)   Setting sample rate to: 8000000
[2022-05-16 22:28:12.774299] (info)   Actual sample rate: 8000000
[2022-05-16 22:28:12.774328] (info)   Tuning to 8.574975e+08
[2022-05-16 22:28:12.774434] (info)   Gain Stage: IFGR supported values: 20 21 22 23 24 25 26 27 28 29 30 31 32 33 34 35 36 37 38 39 40 41 42 43 44 45 46 47 48 49 50 51 52 53 54 55 56 57 58 59 
[2022-05-16 22:28:12.774465] (info)   Gain Stage: RFGR supported values: 0 1 2 3 
[2022-05-16 22:28:12.774481] (info)   Max Frequency: 8.614555e+08
[2022-05-16 22:28:12.774517] (info)   Min Frequency: 8.535195e+08
[2022-05-16 22:28:12.774549] (info)   RFGR Gain set to: 0
[1m[33m[WARNING] Not updating IFGR gain because AGC is enabled[0m
[2022-05-16 22:28:12.774581] (info)   IFGR Gain set to: 50
[2022-05-16 22:28:12.774596] (info)   Auto gain control is ON
[2022-05-16 22:28:12.774609] (info)   Setting antenna to [RX]
[2022-05-16 22:28:12.778253] (info)      P25 Recorder two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 8000000
[2022-05-16 22:28:12.784463] (info)      P25 Recorder ARB - Initial Rate: 8000000 Resampled Rate: 25000 Initial Decimation: 80 ARB Rate: 0.96
```

As you can see, AGC is not disabled properly and the IF/RF gain reductions are not set to the correct values (well, RFGR is but IFGR definitely isn't). IFGR seems to always revert to 50 and RFGR always stays 0. Despite changing both `agc` to `false` and adding `agc_setpoint=0` to the device parameters, AGC remains on which makes my RSP1 unusable for the trunking system I'm monitoring.

_Originally posted by @W3AXL in https://github.com/robotastic/trunk-recorder/issues/422#issuecomment-1128334162_

W3AXL, 2022-05-17:
A little more testing - the heart of the issue is AGC being forced on - `IFGR` can't be set when it's enabled.

According to https://github.com/robotastic/trunk-recorder/issues/422#issuecomment-961987028 AGC must be first enabled then disabled again to properly disable, but it doesn't look like that process got committed anywhere in #546

W3AXL, 2022-05-17:
This is also preventing me from decoding the system I'm tracking. Control channel is decoding fine, but all audio calls are heavily garbled which I'm guessing is due to the overall signals being too low. My previous setup with 3 RTL dongles worked fine on the same antenna.

photounit, 2022-07-21:
As far as the AGC goes, try setting the AGC on and then add another line turning it off. I know that sounds odd, but this is what worked for me when I had the same issue with the BladeRF.  And you have gotten further getting the SDRPlays to work that I have. I have two that I would like to use, but their support for usage in non-SDRPlay software sucks IMO....

tadscottsmith, 2022-07-21:
I wonder if you take line 532 in main.cc: https://github.com/robotastic/trunk-recorder/blob/a30ee308518d478c4fdcd39fab2314717898bd65/trunk-recorder/main.cc#L532

Then slightly modify the ACG code and move it above the other gain settings (prior to line 473) something like this:
```
if(!agc){
    source->set_gain_mode(true);  // Enable AGC before disabling AGC
    source->set_gain_mode(false);
}
else
{
    source->set_gain_mode(true);
}
```
Does that fix the issue? I don't have anything to test it with, but based on your comments it seems like it may just be an order of operations issue.

W3AXL, 2022-07-21:
Sorry for abandoning this issue - life has been busy. You guys are correct - I added lines in the code to toggle AGC on startup and that seems to have resolved the gain settings issue. I'll try and remember to do a PR later today for that change.

Now there's an entirely different issue where higher samplerates (above 2 MHz) are unable to demodulate P25, but I'm still trying to track down whether that's a signal level issue or an issue with the OP25 decimators.

PR #853 (2023-11-11): Fix SoapySDRPlay3 edgecase
Fixes #689 when using https://github.com/pothosware/SoapySDRPlay3. 

I have an SDRPlay device and a [set of Docker images](https://github.com/USA-RedDragon/trunk-recorder-soapysdrplay3) using SoapySDRPlay3 and the newer SDRPlay API. This works great with only this minor code change due to an edgecase with the SoapySDRPlay3 API.

In this case per https://github.com/pothosware/SoapySDRPlay3/blob/534a173996b0ca2a88ff1931950c869d244c90f8/Settings.cpp#L517, the autogain must be disabled before setting the gain to avoid gain not being set.

I've added a new call to `source [... cut]
closed 2023-11-11 by robotastic