TrunkRecorder/trunk-recorder #691: Trunk-Recorder on Ubuntu 22.04

dgwood12, 2022-05-18:
Is there support for Ubuntu 22.04 (64-bit)? I tried installing trunk-recorder on a new install of Ubuntu 22.04 and encountered some errors when running the following command:
cmake ../trunk-recorder

The here's what the command did:

-- Build type not specified: defaulting to release.
-- Using GMP.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found suitable version "1.74.0", minimum required is "1.74.0") found components: date_time program_options system regex thread unit_test_framework 
-- User set python executable /usr/bin/python3
-- Extracting version information from git describe...
-- Using GMP.
-- Checking for module 'sndfile'
--   No package 'sndfile' found
-- Could NOT find SNDFILE (missing: SNDFILE_LIBRARIES SNDFILE_INCLUDE_DIRS) 
-- GnuRadio Version: 200705
-- Found gnuradio-uhd: /usr/include, /usr/lib/x86_64-linux-gnu/libgnuradio-uhd.so
-- Checking for module 'gnuradio-osmosdr'
--   Found gnuradio-osmosdr, version 0.2.0.0
-- gnuradio-osmosdr not found.
-- Found gnuradio-osmosdr: , gnuradio-osmosdr;gnuradio-blocks;gnuradio-runtime;gnuradio-pmt
-- Could NOT find GnuradioOsmosdr (missing: GNURADIO_OSMOSDR_INCLUDE_DIRS) 
-- Configuring Boost C++ Libraries...
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.74.0/BoostConfig.cmake (found version "1.74.0") found components: thread system program_options filesystem log_setup log regex random 
-- AMBE Tone Synthesis Enabled
-- All libraries:
-- Configuring done
CMake Error at CMakeLists.txt:338 (add_executable):
  Target "trunk-recorder" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at lib/op25_repeater/lib/CMakeLists.txt:110 (add_library):
  Target "gnuradio-op25_repeater" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at plugins/openmhz_uploader/CMakeLists.txt:1 (add_library):
  Target "openmhz_uploader" links to target "sndfile::sndfile" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at plugins/stat_socket/CMakeLists.txt:1 (add_library):
  Target "stat_socket" links to target "sndfile::sndfile" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at plugins/broadcastify_uploader/CMakeLists.txt:1 (add_library):
  Target "broadcastify_uploader" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at plugins/unit_script/CMakeLists.txt:1 (add_library):
  Target "unit_script" links to target "sndfile::sndfile" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Error at plugins/rdioscanner_uploader/CMakeLists.txt:1 (add_library):
  Target "rdioscanner_uploader" links to target "sndfile::sndfile" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at plugins/simplestream/CMakeLists.txt:1 (add_library):
  Target "simplestream" links to target "sndfile::sndfile" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

Is there anything I can do to get trunk-recorder going on Ubuntu 22.04 (64-bit)?

dgwood12, 2022-05-27:
Missed the supported version on the readme. Closing out.

closed 2022-05-27 by dgwood12

MaxwellDPS, 2022-05-31:
Hey @dgwood12 can we get a direct link,I cant find this info either and its also affecting @Dygear