TrunkRecorder/trunk-recorder #693: Build fails on Fedora 36 and Arch with /usr/include/spdlog/fmt/fmt.h:24:14: fatal error: spdlog/fmt/bundled/core.h: No such file or directory

mldulaney, 2022-05-20:
In file included from /usr/include/spdlog/common.h:45,
                 from /usr/include/gnuradio/logger.h:36,
                 from /usr/include/gnuradio/custom_lock.h:15,
                 from /usr/include/gnuradio/buffer.h:15,
                 from /usr/include/gnuradio/buffer_double_mapped.h:15,
                 from /usr/include/gnuradio/io_signature.h:17,
                 from /usr/include/gnuradio/basic_block.h:15,
                 from /usr/include/gnuradio/hier_block2.h:15,
                 from /builddir/build/BUILD/trunk-recorder-master/trunk-recorder/recorders/recorder.h:20,
                 from /builddir/build/BUILD/trunk-recorder-master/trunk-recorder/recorders/recorder.cc:1:
/usr/include/spdlog/fmt/fmt.h:24:14: fatal error: spdlog/fmt/bundled/core.h: No such file or directory
   24 | #    include <spdlog/fmt/bundled/core.h>
      |              ^~~~~~~~~~~~~~~~~~~~~~~~~~~

This happens under Fedora and Arch; it seems that the spdlog dev is, shall we say stubborn, and is bundling libraries that they should not be instead of using the installed system fmt library.

robotastic, 2022-05-30:
Hmm...  I am not sure if there is a way to fix this. Those headers are from GNU Radio. Is there a different way to install spdlog?

robotastic, 2022-06-07:
Going to close for now... let me know if this is still an issue

closed 2022-06-07 by robotastic