TrunkRecorder/trunk-recorder #697: Error introduced at c66888e9213c4bfab607ec2ed627b16458faeac4

kb2ear, 2022-06-05:
on the commit above errors started in the cmake on debian 10

root@trunk-recorder:~/trunk-recorder# git reset --hard c66888e9213c4bfab607ec2ed627b16458faeac4
HEAD is now at c66888e Updated from Upstream
root@trunk-recorder:~/trunk-recorder# cmake ../trunk-recorder
-- The CXX compiler identification is GNU 8.3.0
-- The C compiler identification is GNU 8.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.13
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: ANALOG
-- Checking for module 'gnuradio-analog'
--   Found gnuradio-analog, version 3.7.13
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so;/usr/lib/x86_64-linux-gnu/libvolk.so
-- Found GNURADIO_ANALOG: /usr/lib/x86_64-linux-gnu/libgnuradio-analog.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so;/usr/lib/x86_64-linux-gnu/libvolk.so
GNURADIO_ANALOG_FOUND = TRUE
Checking for GNU Radio Module: BLOCKS
-- Checking for module 'gnuradio-blocks'
--   Found gnuradio-blocks, version 3.7.13
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_BLOCKS: /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_BLOCKS_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
-- Checking for module 'gnuradio-digital'
--   Found gnuradio-digital, version 3.7.13
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_DIGITAL: /usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FFT
-- Checking for module 'gnuradio-fft'
--   Found gnuradio-fft, version 3.7.13
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_FFT: /usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.13
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_FILTER: /usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.13
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found GNURADIO_PMT: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so;/usr/lib/x86_64-linux-gnu/liblog4cpp.so
GNURADIO_PMT_FOUND = TRUE
[... 182 lines cut ...]
  but the target was not found.  Perhaps a find_package() call is missing for
  an IMPORTED target, or an ALIAS target is missing?

CMake Error at plugins/rdioscanner_uploader/CMakeLists.txt:1 (add_library):
  Target "rdioscanner_uploader" links to target "gnuradio::gnuradio-pmt" but
  the target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at plugins/simplestream/CMakeLists.txt:1 (add_library):
  Target "simplestream" links to target "gnuradio::gnuradio-runtime" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at plugins/simplestream/CMakeLists.txt:1 (add_library):
  Target "simplestream" links to target "gnuradio::gnuradio-filter" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at plugins/simplestream/CMakeLists.txt:1 (add_library):
  Target "simplestream" links to target "gnuradio::gnuradio-pmt" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

-- Generating done
-- Build files have been written to: /root/trunk-recorder

kb2ear, 2022-06-05:
and on compile:
```
[ 67%] Linking CXX shared library ../../../libgnuradio-op25_repeater.so
/usr/bin/ld: cannot find -lgnuradio::gnuradio-runtime
/usr/bin/ld: cannot find -lgnuradio::gnuradio-filter
/usr/bin/ld: cannot find -lgnuradio::gnuradio-pmt
collect2: error: ld returned 1 exit status
make[2]: *** [lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/build.make:1047: libgnuradio-op25_repeater.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:280: lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

robotastic, 2022-06-07:
oh! This is GR 3.7: `Found gnuradio-runtime, version 3.7.13`
Let me see how I can spin that up to test again.

robotastic, 2022-06-07:
Weird though - it doesn't look like there any real changes to the cmake file.
https://github.com/robotastic/trunk-recorder/commit/c66888e9213c4bfab607ec2ed627b16458faeac4

kb2ear, 2022-06-22:
poke

tadscottsmith, 2022-06-25:
I just confirmed on my RPI, #e1c303a combines but #c6688e does not.

bctrainers, 2022-06-29:
On Debian 10, this same error surfaces. Upgrading [or installing] to Debian 11 resolved this issue. Seems to be fully related to gnuradio utilities/packages.

robotastic, 2022-07-02:
OK! I think I got this fixed. There was some CMAKE weirdness that was introduced when support for GNU Radio 3.10 was introduced. I tested with Debian 10, which is using GR 3.7. Open this back up if there are still problems. Thanks everyone for all the clues in hunting this down!!!

https://github.com/robotastic/trunk-recorder/commit/68d3c71baacb8a58c778a8b125711ab5b09fa9e1

kb2ear, 2022-07-02:
OK so now we get:
```
root@trunk-recorder:~/trunk-build# make
[  0%] Checking the git repository for changes...
[  0%] Built target check_git
[  1%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/call_impl.cc.o
In file included from /root/trunk-recorder/trunk-recorder/systems/p25_trunking.h:59,
                 from /root/trunk-recorder/trunk-recorder/systems/system_impl.h:9,
                 from /root/trunk-recorder/trunk-recorder/call_impl.h:18,
                 from /root/trunk-recorder/trunk-recorder/call_impl.cc:1:
/root/trunk-recorder/trunk-recorder/systems/../gr_blocks/rms_agc.h:8:10: fatal error: gnuradio/blocks/multiply_const.h: No such file or directory
 #include <gnuradio/blocks/multiply_const.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:76: CMakeFiles/trunk_recorder_library.dir/trunk-recorder/call_impl.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/trunk_recorder_library.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

kb2ear, 2022-07-02:
As best I can tell this file does not exist in GNUradio 3.7

robotastic, 2022-07-03:
I think this is really fixed now :)

closed 2022-07-03 by robotastic