TrunkRecorder/trunk-recorder #702: Error After minDuration Transmission Removal

dgwood12, 2022-06-13:
I have a couple of channels that I can't quite get the squelch right on that have a 0 to half a second static click sent out every so often. To combat this, I decided to implement the minDuration and minTransmissionDuration properties in my system section. When trunk recorder eliminates the transmissions that are below the minimum duration threshold, it is followed with an error saying No Transmissions were Recorded! 

Correct me if I an wrong but I do not believe there should be an error that follows as the transmission not being recorded is the intended result.

Here's an example of the error:

[2022-06-13 00:53:13.006652] (info)   [PCFD]	5C	TG:          4	Freq: 154.175000 MHz	Removing transmission less than 0.5 seconds. Actual length: 0.24725.
[2022-06-13 00:53:13.006742] (error)   [PCFD]	5C	TG:          4	 Freq: 1.54175e+08	No Transmission were recorded!

Should the error be fixed to not generate in these instances?

Thanks

tadscottsmith, 2022-06-15:
You will get that error if the transmission that was removed was the only transmission for the call. However, that error also can happen if for some other reason (signal, etc.) no transmissions were recorded, so it's probably good to keep the error there.

tadscottsmith, 2022-11-21:
@robotastic I created a PR #735 that addresses this issue.

If no transmissions were recorded, it will still be logged as an error:
```
[2022-11-21 16:19:15.057531] (error)   [ramsey] 3034C   TG:      11700   Freq: 8.53112e+08      No Transmissions were recorded!
```
However, if all transmissions were removed because they did not meet the minimum transmission duration, it will be logged as informational:
```
[2022-11-21 16:22:18.073246] (info)   [ramsey]  3178C   TG:      11562   Freq: 8.52688e+08      No Transmissions were recorded! 2 tranmissions less than 0.35 seconds were removed.
```

tadscottsmith, 2022-11-21:
Just noticed I misspelled transmissions in  "_tranmissions less than_" if you wanted to fix that up before committing. Thanks!

tadscottsmith, 2022-12-15:
@dgwood12 can you check this out and close accordingly?

closed 2023-04-09 by dgwood12