TrunkRecorder/trunk-recorder #703: multi sites multi bandplan Is it possible?

LoveScanner, 2022-06-14:
I need to set 3 sites in different frequency range (3 dongle)
On each site I need to set up a custom bandplan
how do I do it?
From the config I see that only one custom bandplan can be defined in each config file

Is it possible at all?
Anyone have an example?

HarrisonWilson, 2022-06-19:
So I believe the confusion here would be with the terminology. Although there is discussion of "multiple systems" what is actually meant is "multiple sites" since the idea is typically for each system you would most likely only have one site for a system in range. If you look at OpenMHz or Broadcastify Calls, you will note that each "system" may only monitor a site (or simulcast, which is several physical sites appearing as one logical site to the subscriber).

What I would do is set up your dongle(s) to cover the desired frequency range, than if you have three sites in range, say "North Simulcast", "East Simulcast", and "West Simulcast" each one would be its own system even though the SYS ID is the same, each one has a different set of control channels and band plans.

Hopefully that makes sense. Use the [multi-system example](https://github.com/robotastic/trunk-recorder/blob/master/examples/config-multi-system.json) in the repository as reference.

LoveScanner, 2022-06-19:
> So I believe the confusion here would be with the terminology. Although there is discussion of "multiple systems" what is actually meant is "multiple sites" since the idea is typically for each [...]

Thanks for the information I will try it soon

Just for the sake of understanding,
trunk-recoder acts like sdrtrunk
It basically records all the Talkgroups at once and does not act like a scanner that records only one channel at a time

I understand it right?

HarrisonWilson, 2022-06-20:
Correct. 

So for your case what could happen is if a talkgroup appears on two sites at once it will be recorded twice. All controls channels of all systems (or sites) will be continuously monitored.

That said, as long as you have enough recorders (part of the reviver configuration) to cover all the potential traffic you will be able to record everything simultaneously. For example, if you only have 10 recorders configured and the system you are listening to could potentially have 12 conversations simultaneously the last two conversations to start would not be recorded. 

The downside to more recorders is more CPU requirements, hence why you would only set up what you need or can support.

LoveScanner, 2022-07-26:
If my control channel and the speech channel are in the sample rate range that my dongle supports, does that mean that one dongle is enough for me? (SmartZone Motorola)

taclane, 2022-07-26:
> 1. If my control channel and the speech channel are in the sample rate range that my dongle supports, does that mean that one dongle is enough for me? (SmartZone Motorola Trunking)
> 2. Does the trunk-recorder know how to record several channels at the same time or does it record one at a time?

1. Yes. You only need as many devices as you need bandwidth.  If one covers all the channels you need, then one will do.  Though, if the frequencies you are monitoring are close to either edge, reception may be a little degraded.
2. In a trunked system, Trunk Recorder is only monitoring the Control Channel full-time.  When a voice transmission happens it will assign an idle recorder to cover it.  Depending on the system type either `digitalRecorders` or `analogRecorders` is set in the config file to have that many recorders standing by for transmissions.  It can be a little taxing on system resources if you have too many; some recon or trial-and-error may be required depending on how active the system is.

closed 2024-12-14 by robotastic