TrunkRecorder/trunk-recorder #706: AUR Arch include spdlog/fmt/bundled/core.h

NotSBad, 2022-06-23:
When installing a package through Aur, the following error occurs:

`usr/include/spdlog/fmt/fmt.h:24:14: fatal error: spdlog/fmt/bundled/core.h: No such file or directory 24 | # include spdlog/fmt/bundled/core.h`

Please tell me how to fix it?
spdlog-1.10.0-1 installed
Thanks

toddejohnson, 2022-06-28:
This looks similar to #693. Testing in [trunk-recorder-git](https://aur.archlinux.org/packages/trunk-recorder-git) appears to get past this with r1827.8f0dc53-1 although I haven't tested it past build.

mldulaney, 2022-07-05:
Yeah, this is that problem.

robotastic, 2022-07-06:
Is this something I can fix in Trunk Recorder or does it have to do with GNU Radio and the libraries it is bundling?

mldulaney, 2022-07-27:
This seems to have been resolved in fedora and arch by the spdlog packagers.

toddejohnson, 2022-07-27:
From my quick look in [arch they just rebuilt](https://github.com/archlinux/svntogit-community/commits/packages/spdlog/trunk/PKGBUILD)  and [Fedora did the same but specified it was for an fmt library update](https://src.fedoraproject.org/rpms/spdlog/c/be8cd7fe94ddb22fb79e51350dbb8fb599f232f1?branch=rawhide). 

Still strange that with those updates I can build the -git package but not release. 

Bisecting the build it looks like #698 likely dd25c29 fixed this. Since there isn't a release including this it is causing issues building the released version.

toddejohnson, 2022-07-28:
I can confirm #698 helps with this issue building v4.3.2 on arch. Then it needed #682 too for GR 3.10. A bit difficult to patch after some renames(/lib) and others. 

I'll release an AUR package [4.3.2-2](https://aur.archlinux.org/packages/trunk-recorder) with the patches but it will be untested. Looks like there are great devs fixing these issues and a bunch of new features we can be exited for a new release. The AUR package [trunk-recorder-git](https://aur.archlinux.org/packages/trunk-recorder-git) is an alternative that is also working and has nice new features and fixes although not exactly released.

closed 2024-12-14 by robotastic