TrunkRecorder/trunk-recorder #707: cannot build from git in portage/emerge

ZeroChaos-, 2022-07-01:
I've had a "live ebuild" for TR for a while now, basically it automatically pulls from git within gentoo's package manager, and builds/installs the latest rev.  It's super helpful for testing, as it runs everything through all the qa scripts, etc.  Unfortunately, when version information was added to the build system, it breaks the way gentoo builds.
```
[147/159] : && /usr/bin/x86_64-pc-linux-gnu-g++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -pthread -Wno-narrowing -fvisibility=hidden -fPIC -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic CMakeFiles/trunk-recorder.dir/trunk-recorder/main.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/formatter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/source.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_conventional.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/system_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/debug_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/sigmf_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/analog_recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/dmr_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_fsk4_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_qpsk_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/csv_helper.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroup.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroups.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tag.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tags.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/plugin_manager/plugin_manager.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_concluder/call_concluder.cc.o CMakeFiles/trunk-recorder.dir/lib/lfsr/lfsr.cxx.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/freq_xlating_fft_filter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/transmission_sink.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/fsync_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/mdc_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/star_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/signal_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/tps_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoder_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/plugin_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/wavfile_gr3.8.cc.o -o trunk-recorder  -Wl,-rpath,/var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999_build:  libgit.so  libtrunk-recorder.a  libgnuradio-op25_repeater.so  -ldl  -lssl  -lcrypto  -lcurl  /usr/lib64/libboost_log_setup.so.1.79.0  /usr/lib64/libboost_log.so.1.79.0  /usr/lib64/libboost_random.so.1.79.0  -lgnuradio-uhd  -luhd  -lgnuradio-osmosdr  /usr/lib64/libgnuradio-digital.so.3.8.5.0  /usr/lib64/libboost_chrono.so.1.79.0  -lgnuradio-filter  /usr/lib64/libgnuradio-analog.so.3.8.5.0  /usr/lib64/libgnuradio-filter.so.3.8.5.0  /usr/lib64/libgnuradio-blocks.so.3.8.5.0  /usr/lib64/libgnuradio-fft.so.3.8.5.0  -lfftw3f  -lfftw3f_threads  /usr/lib64/libgnuradio-runtime.so.3.8.5.0  /usr/lib64/libboost_system.so.1.79.0  /usr/lib64/libboost_program_options.so.1.79.0  /usr/lib64/libboost_filesystem.so.1.79.0  /usr/lib64/libboost_atomic.so.1.79.0  /usr/lib64/libboost_regex.so.1.79.0  /usr/lib64/libgnuradio-pmt.so.3.8.5.0  /usr/lib64/libboost_thread.so.1.79.0  -llog4cpp  -lgmpxx  -lgmp  -lrt  /usr/lib64/libvolk.so.2.5.0  -ldl  -lm && :
FAILED: trunk-recorder 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -pthread -Wno-narrowing -fvisibility=hidden -fPIC -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic CMakeFiles/trunk-recorder.dir/trunk-recorder/main.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/formatter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/source.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_conventional.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/system_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/debug_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/sigmf_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/analog_recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/dmr_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_fsk4_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_qpsk_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/csv_helper.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroup.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroups.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tag.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tags.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/plugin_manager/plugin_manager.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_concluder/call_concluder.cc.o CMakeFiles/trunk-recorder.dir/lib/lfsr/lfsr.cxx.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/freq_xlating_fft_filter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/transmission_sink.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/fsync_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/mdc_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/star_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/signal_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/tps_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoder_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/plugin_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/wavfile_gr3.8.cc.o -o trunk-recorder  -Wl,-rpath,/var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999_build:  libgit.so  libtrunk-recorder.a  libgnuradio-op25_repeater.so  -ldl  -lssl  -lcrypto  -lcurl  /usr/lib64/libboost_log_setup.so.1.79.0  /usr/lib64/libboost_log.so.1.79.0  /usr/lib64/libboost_random.so.1.79.0  -lgnuradio-uhd  -luhd  -lgnuradio-osmosdr  /usr/lib64/libgnuradio-digital.so.3.8.5.0  /usr/lib64/libboost_chrono.so.1.79.0  -lgnuradio-filter  /usr/lib64/libgnuradio-analog.so.3.8.5.0  /usr/lib64/libgnuradio-filter.so.3.8.5.0  /usr/lib64/libgnuradio-blocks.so.3.8.5.0  /usr/lib64/libgnuradio-fft.so.3.8.5.0  -lfftw3f  -lfftw3f_threads  /usr/lib64/libgnuradio-runtime.so.3.8.5.0  /usr/lib64/libboost_system.so.1.79.0  /usr/lib64/libboost_program_options.so.1.79.0  /usr/lib64/libboost_filesystem.so.1.79.0  /usr/lib64/libboost_atomic.so.1.79.0  /usr/lib64/libboost_regex.so.1.79. [... cut ...]
C++             -Os -march=native -mtune=native -pipe -frecord-gcc-switches -pthread -Wno-narrowing -fvisibility=hidden -fPIC
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Module          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0
Shared          -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0

-- Configuring done
-- Generating done
-- Build files have been written to: /var/tmp/portage/net-wireless/trunk-recorder-9999/work/trunk-recorder-9999_build
```

taclane, 2022-07-18:
Just tested this on both a fresh master.zip and a git pull; comment out the below lines in `git_watcher.cmake`:
https://github.com/robotastic/trunk-recorder/blob/a30ee308518d478c4fdcd39fab2314717898bd65/git_watcher.cmake#L137-L140
and it will build.  T-R will fall back on the default version strings if queried from a snapshot build:
```
# ./trunk-recorder --version
Trunk-Recorder: 4.3.2
WARN: failed to get the current git state. Is this a git repo?
```
Or display git stats from a git build:
```
# ./trunk-recorder --version
Trunk-Recorder: 4.3.2
        WARN: there were uncommitted changes at build-time.
        commit a30ee308518d478c4fdcd39fab2314717898bd65 (master)
        describe 2.1.2-1460-ga30ee30
        Author: Luke Berndt <lukekb@gmail.com>
        Date: 2022-07-16 09:42:05 -0400

        updated costas_alpha to match OP25
```
@rabarar , since you helped get this going with #688, do you have any other suggestions?

rabarar, 2022-07-18:
What version of git are you using?

ZeroChaos-, 2022-07-18:
It fails because the .git directory is missing, I am very confident that is unrelated to the git software version.

ZeroChaos-, 2022-07-18:
That said, looks like today it is 2.35.1

taclane, 2022-07-18:
> What version of git are you using?
> […](#)

I tested the above with `git version 2.37.1` on MacOS/homebrew.  As noted previously, building from a [source zip](https://github.com/robotastic/trunk-recorder/archive/refs/heads/master.zip) will fail because it lacks a live `.git` directory.

Per the comments just above the lines I quoted in git_watcher.cmake, commenting them out will allow the compile to continue with default values if the source is a tagged release or tarball, or anything other than a git pull.
```
        # Issue 26: git info not properly set
        #
        # Check if we should fail if any of the exit codes are non-zero.
        # Most methods have a fall-back default value that's used in case of non-zero
        # exit codes. If you're feeling risky, disable this safety check and use
        # those default values.
```
I haven't tested is what happens if you dont even have git installed at all.  I don't know what would happen in that event.

rabarar, 2022-07-18:
Ah, I see… let me look and see what I can do to conditionally compile when there is no repo…

ZeroChaos-, 2022-07-18:
Being able to inject a version by passing it in directly via cmake would be nice as well.

rabarar, 2022-07-19:
The version is pulled out of the code in addition to the git status.

The solution i think is to modify line 139 of git_watcher.cmake and revise the output on line 53 of git.cc <http://git.cc/>.in to indicate no git repository is availble. Or I can delete it altogether and solely print out the version numbers extracted from the source. like the test below:

```
% ./trunk-recorder --version
Trunk-Recorder: 4.3.2

```

Thoughts/preferences? Let me know and I’ll do a PR

ZeroChaos-, 2022-07-19:
Despite it breaking my builds, I thought the idea was great.  I really just want a way to pass in the information for when auto extraction fails.  Also, it should fall back to the static version tag if the git extraction fails and nothing is passed in via cmake.

rabarar, 2022-07-19:
I didn’t realize anyone would be building from a non-repo so that’s why it was a fatal error. What information would one pass in if the git repo ins’t there??  Not sure I follow passing in git info from a non-git repo. 

I think the solution is to go silent, build and just not output anything related to git stats on the ‘—version’ output.

ZeroChaos-, 2022-07-19:
It's the way the Gentoo package manager works.  It is building a git version but it doesn't keep the .git directory while building.  So I have the info, but no way to pass it in.

ZeroChaos-, 2022-07-19:
It is also a fatal error for building from a snapshot, or a tag, so if another release ever happens, a lot of people will be compiling without a .git directory.

ZeroChaos-, 2022-08-23:
The git related stuff of this was already fixed, I'm going to open a new bug to cover the other part of the issue which was confused in this issue originally.

closed 2022-08-23 by ZeroChaos-