TrunkRecorder/trunk-recorder #711: Recorder number incorrect on analog conventional

wx-jc, 2022-07-04:
On conventional, the printout that occurs when you start TR shows too many recorders.  It appears to start with the total number you have (in my example, 6) and increment from there. 

[image]

ZeroChaos-, 2022-07-04:
This is documented as well. https://github.com/robotastic/trunk-recorder/blob/master/docs/CONFIGURE.md

taclane, 2022-07-04:
Yup.  All recorders are in one big list, and conventionals get numbered after any trunked recorders have started up.  If your config has `"analogRecorders": 6` (used by trunked only), that could explain why the recorder numbers have shifted higher than expected.  Or if you have a mix of six trunked recorders for another system, it would have the same overall effect on numbering.

wx-jc, 2022-07-04:
Thanks for the tip!  That is indeed confusing.  I'm using a json that has been modified over the years.  I don't think that note in the link you sent was there when I first got involved in this years ago, so missed it.

closed 2022-07-04 by wx-jc