TrunkRecorder/trunk-recorder #713: Build fails when cmake -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG

mldulaney, 2022-07-06:
The cmake flag -DCMAKE_CXX_FLAGS_RELEASE:STRING=-DNDEBUG is a standard flag when building packages for Fedora; including it causes the build to fail to link due to multiple definition errors.

/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::~selector_impl()'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `virtual thunk to gr::blocks::selector_impl::~selector_impl()'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::~selector_impl()'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `virtual thunk to gr::blocks::selector_impl::~selector_impl()'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::~selector_impl()'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::forecast(int, std::vector<int, std::allocator<int> >&)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `virtual thunk to gr::blocks::selector_impl::forecast(int, std::vector<int, std::allocator<int> >&)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::set_input_index(unsigned int)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::set_output_index(unsigned int)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `virtual thunk to gr::blocks::selector_impl::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::handle_enable(std::shared_ptr<pmt::pmt_base>)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::check_topology(int, int)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `virtual thunk to gr::blocks::selector_impl::check_topology(int, int)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::setup_rpc()'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `virtual thunk to gr::blocks::selector_impl::setup_rpc()'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::selector_impl(unsigned long, unsigned int, unsigned int)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector_impl::selector_impl(unsigned long, unsigned int, unsigned int)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here
/usr/bin/ld: selector_impl.cc.o (symbol from plugin): in function `gr::blocks::selector_impl::~selector_impl()':
(.text+0x0): multiple definition of `gr::blocks::selector::make(unsigned long, unsigned int, unsigned int)'; CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o (symbol from plugin):(.text+0x0): first defined here

mldulaney, 2022-08-06:
beep?

robotastic, 2024-12-14:
Happy to take a PR if you figure out a fix for this - closing for now

closed 2024-12-14 by robotastic