TrunkRecorder/trunk-recorder #714: Can a SIngle Source/SDR Supply Two Systems?

Dewey3, 2022-07-16:
I am in range of JNCR 580A0-006 (Silver Hill) and 580A0-015 (Indian Head) and experimenting with setting up a RPi for the two systems.  Since the frequencies for the JNCR 580A0-015 system will actually fit "inside" of the frequencies for the JNCR 580A0-006 system`*`, and both systems total only 9 frequencies, will I have any problems if I run both "systems" objects from the same "source" object in the json file, or should I dedicate a second source/SDR to cover the second system?  Right now, I have the json file setup for a single source and two systems, and while it's not throwing any errors, I wonder if it's actually monitoring both systems.  One potential problem I am seeing is that the recording subdirectoies for both systems are populating with transmissions, but neither one seems to be feeding Rdio via the single source json.

`* :`
580A0-006 low - 380.275  /  580A0-006 high - 381.7375  /  5 frequencies total
580A0-015 low - 380.450  /  580A0-015 high - 381.7000  /  4 frequencies total

```
"sources": [{
  "center": 381006250,
  "rate": 1920000,
  "error": 0,
  "gain": 29.7,
  "digitalRecorders": 10,
  "driver": "osmosdr",
  "device": "rtl=20160923",
  "ppm": 0
```

tadscottsmith, 2022-07-18:
From a trunk-recorder perspective, it should be fine as long as you can receive both systems with the the same signal settings (gain). I'm feeding two systems spread across the same sources using the plugin with something similar to this:
``` "plugins": [
         {
          "name": "rdioscanner_uploader",
          "library": "librdioscanner_uploader.so",
          "server": "https://rdio-host",
          "systems": [
           {
                  "shortName": "system1",
                  "apiKey": "APIKEY",
                  "systemId": 1
           },
           {
                  "shortName": "system2",
                  "apiKey": "APIKEY",
                  "systemId": 1
           }]
          }
  ],
  ```
I've never used the dirwatch option, so I am not sure if rdio supports that method for multiple systems.

Dewey3, 2022-07-18:
Thanks @tadscottsmith .  I did finally figure out how to incorporate two systems into the plugin just as you outlined.  Just an FYI for anyone else who may be following the thread, I had it working locally with Rdio's `dirwatch` by setting dirwatch to the lower directory level, so instead of having dirwatch monitoring the single top level `/recordings` directory. I had two dirwatch(es), one set to `/recordings/580A0-006` and the other set to `/recordings/580A0-015`.

Again, thanks for all your help.

closed 2022-07-18 by Dewey3