TrunkRecorder/trunk-recorder #719: unabled to build shared

ZeroChaos-, 2022-08-23:
`cmake -DBUILD_SHARED_LIBS=ON` fails but building static works.  I'm not an expert, but I'm guessing something gets stripped out of libgit.so which isn't stripped out when it's libgit.a

```
[2/8] : && /usr/bin/c++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -pthread -Wno-narrowing -fvisibility=hidden -fPIC -O3 -DNDEBUG -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic CMakeFiles/trunk-recorder.dir/trunk-recorder/main.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/formatter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/source.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_conventional.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/system_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/debug_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/sigmf_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/analog_recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/dmr_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_fsk4_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_qpsk_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/csv_helper.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroup.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroups.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tag.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tags.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/plugin_manager/plugin_manager.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_concluder/call_concluder.cc.o CMakeFiles/trunk-recorder.dir/lib/lfsr/lfsr.cxx.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/freq_xlating_fft_filter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/transmission_sink.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/fsync_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/mdc_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/star_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/signal_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/tps_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoder_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/plugin_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/wavfile_gr3.8.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/rms_agc.cc.o -o trunk-recorder  -Wl,-rpath,/home/zero/development/trunk-recorder/build:  libgit.so  libtrunk-recorder.a  libgnuradio-op25_repeater.so  -ldl  -lssl  -lcrypto  -lcurl  /usr/lib64/libboost_log_setup.so.1.79.0  /usr/lib64/libboost_log.so.1.79.0  /usr/lib64/libboost_random.so.1.79.0  -lgnuradio-uhd  -luhd  -lgnuradio-osmosdr  /usr/lib64/libgnuradio-digital.so.3.8.5.0  /usr/lib64/libboost_chrono.so.1.79.0  -lgnuradio-filter  /usr/lib64/libgnuradio-analog.so.3.8.5.0  /usr/lib64/libgnuradio-filter.so.3.8.5.0  /usr/lib64/libgnuradio-blocks.so.3.8.5.0  /usr/lib64/libgnuradio-fft.so.3.8.5.0  -lfftw3f  -lfftw3f_threads  /usr/lib64/libgnuradio-runtime.so.3.8.5.0  /usr/lib64/libboost_system.so.1.79.0  /usr/lib64/libboost_program_options.so.1.79.0  /usr/lib64/libboost_filesystem.so.1.79.0  /usr/lib64/libboost_atomic.so.1.79.0  /usr/lib64/libboost_regex.so.1.79.0  /usr/lib64/libgnuradio-pmt.so.3.8.5.0  /usr/lib64/libboost_thread.so.1.79.0  -llog4cpp  -lgmpxx  -lgmp  -lrt  /usr/lib64/libvolk.so.2.5.0  -ldl  -lm && :
FAILED: trunk-recorder 
: && /usr/bin/c++ -Os -march=native -mtune=native -pipe -frecord-gcc-switches -pthread -Wno-narrowing -fvisibility=hidden -fPIC -O3 -DNDEBUG -Wl,-O1 -Wl,--as-needed -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic CMakeFiles/trunk-recorder.dir/trunk-recorder/main.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/formatter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/source.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_conventional.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_trunking.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/p25_parser.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/smartnet_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/systems/system_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/debug_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/sigmf_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/analog_recorder.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/dmr_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_fsk4_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_qpsk_demod.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/recorders/p25_recorder_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/csv_helper.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroup.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/talkgroups.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tag.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/unit_tags.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/plugin_manager/plugin_manager.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/call_concluder/call_concluder.cc.o CMakeFiles/trunk-recorder.dir/lib/lfsr/lfsr.cxx.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/freq_xlating_fft_filter.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/transmission_sink.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/fsync_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/mdc_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/star_decode.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/signal_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoders/tps_decoder_sink_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/decoder_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/plugin_wrapper_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/selector_impl.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/wavfile_gr3.8.cc.o CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/rms_agc.cc.o -o trunk-recorder  -Wl,-rpath,/home/zero/development/trunk-recorder/build:  libgit.so  libtrunk-recorder.a  libgnuradio-op25_repeater.so  -ldl  -lssl  -lcrypto  -lcurl  /usr/lib64/libboost_log_setup.so.1.79.0  /usr/lib64/libboost_log.so.1.79.0  /usr/lib64/libboost_random.so.1.79.0  -lgnuradio-uhd  -luhd  -lgnuradio-osmosdr  /usr/lib64/libgnuradio-digital.so.3.8.5.0  /usr/lib64/libboost_chrono.so.1.79.0  -lgnuradio-filter  /usr/lib64/libgnuradio-analog.so.3.8.5.0  /usr/lib64/libgnuradio-filter.so.3.8.5.0  /usr/lib64/libgnuradio-blocks.so.3.8.5.0  /usr/lib64/libgnuradio-fft.so.3.8.5.0  -lfftw3f  -lfftw3f_threads  /usr/lib64/libgnuradio-runtime.so.3.8.5.0  /usr/lib64/libboost_system.so.1.79.0  /usr/lib64/libboost_program_options.so.1.79.0  /usr/lib64/libboost_filesystem.so.1.79.0  /usr/lib64/libboost_atomic.so.1.79.0  /usr/lib64/libboost_regex.so.1.79.0  /usr/lib64/libgnuradio-pmt.so.3.8.5.0  /usr/lib64/libboost_thread.so.1.79.0  -llog4cpp  -lgmpxx  -lgmp  -lrt  /usr/lib64/libvolk.so.2.5.0  -ldl  -lm && :
/usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: CMakeFiles/trunk-recorder.dir/trunk-recorder/main.cc.o: in function `main':
main.cc:(.text.startup+0x68d): undefined reference to `GitMetadata::VersionInfo()'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

robotastic, 2022-08-24:
Looks like it could be related to this: https://github.com/diversario/node-ssdp/issues/115

ZeroChaos-, 2022-08-25:
For whatever reason, gentoo builds everything with -DBUILD_SHARED_LIBS=ON, and this fixes building with that parameter.  It may not be enough if at some point trunk-recorder wants to actually be a shared library, but for now it seems to work correctly.

closed 2022-08-27 by robotastic