TrunkRecorder/trunk-recorder #723: P25 Parse error

stevegame, 2022-09-03:
I was trying to use the trunked-recorder on my laptop with my sdr
i want to use my **rafael tuner**

-----
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Found Rafael Micro R820T tuner
-------------->>Using device #0 Realtek RTL2838UHIDIR SN: 00000001
--------------->>Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2022-09-03 02:45:55.276783] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2022-09-03 02:45:55.276860] (info)   Setting sample rate to: 8000000
Invalid sample rate: 8000000 Hz
[2022-09-03 02:45:55.276965] (info)   Actual sample rate: 1024000
[2022-09-03 02:45:55.277010] (info)   Tuning to 8.570000e+08
[2022-09-03 02:45:55.329850] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6 
[2022-09-03 02:45:55.329934] (info)   Max Frequency: 8.609680e+08
[2022-09-03 02:45:55.330008] (info)   Min Frequency: 8.530320e+08
[2022-09-03 02:45:55.356569] (info)   Gain set to: 40.2
[2022-09-03 02:45:55.356686] (info)   Auto gain control is OFF
[2022-09-03 02:45:55.356739] (info)   Setting antenna to [RX]
[2022-09-03 02:45:55.357523] (info)   	 P25 Recorder two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 8000000
[2022-09-03 02:45:55.401456] (info)   	 P25 Recorder ARB - Initial Rate: 8000000 Resampled Rate: 25000 Initial Decimation: 80 ARB Rate: 0.96 SPS: 5

---
I think these are both the same and I'm confused why it says one found and then the other used
But the I think It causes this problem 

---------

PLUGINS
-------------------------------------
[2022-09-03 02:45:55.426184] (info)   Setting up plugin -  Name: openmhz_uploader	 Library file: libopenmhz_uploader.so
[2022-09-03 02:45:55.437056] (info)   Setting up plugin -  Name: broadcastify_uploader	 Library file: libbroadcastify_uploader.so
[2022-09-03 02:45:55.438320] (info)   Setting up plugin -  Name: unit_script	 Library file: libunit_script.so
[2022-09-03 02:45:55.439136] (info)   Setting up plugin -  Name: stat_socket	 Library file: libstat_socket.so
[2022-09-03 02:45:55.441894] (info)   No plugins configured
[2022-09-03 02:45:55.442038] (info)   

[2022-09-03 02:45:55.442102] (info)   [sys_1]	Started with Control Channel: 8.554625e+08
[2022-09-03 02:45:55.442655] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 8000000
[2022-09-03 02:45:55.449604] (info)   	 P25 Trunking ARB - Initial Rate: 8000000 Resampled Rate: 25000 Initial Decimation: 80 System Rate: 24000 ARB Rate: 0.96
Allocating 15 zero-copy buffers
----------------->>[2022-09-03 02:45:56.866904] (error)   P25 Parse error, s:  s0:  s1: d shift: 0 nac: 100 type: 65535 Len: 0
[2022-09-03 02:45:58.011193] (error)   [sys_1]	There is only one control channel defined
[2022-09-03 02:45:58.011309] (error)   [sys_1]	 Control Channel Message Decode Rate: 0.333333/sec, count:  1
----------------->>[2022-09-03 02:45:58.021562] (error)   P25 Parse error, s:  s0:  s1: d shift: 0 nac: 100 type: 65535 Len: 0
------------------>>[2022-09-03 02:45:59.307264] (error)   P25 Parse error, s:  s0:  s1: d shift: 0 nac: 100 type: 65535 Len: 0

-----------

Please help me I haven seen anyone have this problem before.

Thank you (:

wmg819, 2022-09-07:
I sometimes see that error when my rtl tuner's ppm or error rate is off and can't fully decode the control channel due to being slightly off frequency. Have you used Grqx or even SDR ++ to determine what your tuners error rate is? Once you determine that include it in your config .json file using "ppm": (put error rate number here without brackets) and make sure to put a comma after the number. This line can be inserted between the rate and gain lines of the json file.

aaknitt, 2022-12-23:
I'm running into this same thing with certain SDRs.  I think these are the key lines in the console output:

`[2022-09-03 02:45:55.276860] (info) Setting sample rate to: 8000000
Invalid sample rate: 8000000 Hz
[2022-09-03 02:45:55.276965] (info) Actual sample rate: 1024000
[2022-09-03 02:45:55.357523] (info) P25 Recorder two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 8000000`

The config.json file is specifying a sample rate that the SDR hardware doesn't support.  trunk-recorder appears to recognize this and sets the SDR to a sample rate that the hardware does support.  However, it doesn't change its own internal sample rate to match what the hardware is now set to, and therefore it can't decode anything.

Two changes would be helpful:
1) If trunk-recorder recognizes that the sample rate of the hardware is different than what's in the config.json file, it should use the new hardware sample rate itself rather than using the config.json value
2) Get these error messages into the log.txt file...right now they don't show up there, but only in the console output.