TrunkRecorder/trunk-recorder #730: need config option to make some warnings fatal

ZeroChaos-, 2022-10-06:
I spent a few hours today wondering why nothing works, and the results are basically here:
```
[2022-10-06 16:34:30.038992] (info)   MIX Gain: 0
[2022-10-06 16:34:30.038996] (info)   AMP Gain: 0
[2022-10-06 16:34:30.039001] (info)   VGA Gain: 0
[2022-10-06 16:34:30.039005] (info)   VGA1 Gain: 0
[2022-10-06 16:34:30.039010] (info)   VGA2 Gain: 0
[2022-10-06 16:34:30.039014] (info)   Idle Silence: false
[2022-10-06 16:34:30.039018] (info)   Digital Recorders: 0
[2022-10-06 16:34:30.039023] (info)   Debug Recorder: false
[2022-10-06 16:34:30.039028] (info)   SigMF Recorders: 0
[2022-10-06 16:34:30.039032] (info)   Analog Recorders: 0
[2022-10-06 16:34:30.039038] (info)   Source Device: rtl=5,buflen=32768
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.3.0
built-in source types: file rtl rtl_tcp uhd hackrf bladerf airspy soapy 
Using device #5 Realtek RTL2838UHIDIR SN: 00000001
Using 15 buffers of size 32768.
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2022-10-06 16:34:30.848184] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2022-10-06 16:34:30.848206] (info)   Setting sample rate to: 2048000
[R82XX] PLL not locked!
[2022-10-06 16:34:30.971425] (info)   Actual sample rate: 2048000
[2022-10-06 16:34:30.971465] (info)   Tuning to 4.687000e+08
[2022-10-06 16:34:31.023757] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6 
[2022-10-06 16:34:31.023790] (info)   Max Frequency: 4.696600e+08
[2022-10-06 16:34:31.023801] (info)   Min Frequency: 4.677400e+08
[2022-10-06 16:34:31.050676] (info)   Gain set to: 40.2
[2022-10-06 16:34:31.050712] (info)   Auto gain control is OFF
[2022-10-06 16:34:31.050718] (info)   Setting antenna to [RX]
[2022-10-06 16:34:31.050724] (info)   
-------------------------------------

[2022-10-06 16:34:31.050729] (info)   

-------------------------------------
PLUGINS
-------------------------------------
[2022-10-06 16:34:31.050734] (info)   Setting up plugin -  Name: openmhz_uploader	 Library file: libopenmhz_uploader.so
[2022-10-06 16:34:31.052423] (info)   Setting up plugin -  Name: broadcastify_uploader	 Library file: libbroadcastify_uploader.so
[2022-10-06 16:34:31.052700] (info)   Setting up plugin -  Name: unit_script	 Library file: libunit_script.so
[2022-10-06 16:34:31.052911] (info)   Setting up plugin -  Name: stat_socket	 Library file: libstat_socket.so
[2022-10-06 16:34:31.053757] (info)   No plugins configured
[2022-10-06 16:34:31.053784] (info)   
```

See it?  No?  Me either.  SIX screens of text earlier, you see some really useful stuff

```
[2022-10-06 16:34:24.884105] (info)   Loading Talkgroups...
[2022-10-06 16:34:24.884122] (error)   Malformed channel entry at line 1. Found: 6 Expected 7 or 8
[2022-10-06 16:34:24.884135] (error)   Malformed channel entry at line 2. Found: 6 Expected 7 or 8
[2022-10-06 16:34:24.884141] (error)   Warning: skipped 2 of 2 channel entries! Invalid format?
[2022-10-06 16:34:24.884145] (error)   The format is very particular. See https://github.com/robotastic/trunk-recorder for example input.
```

There are TWO lines in that channelfile, and both are ignored.  Yet tr is happily "running" and doing nothing.

Other similar messages show up as they are read, which means if the last channel added fails you see it, but if the first fails and 30 load below it you have no chance to see the message.

```
[2022-10-06 16:38:53.606695] (error)   [dmr]	 Unable to find a source for this conventional channel! Channel not added: 4.765000e+08 Talkgroup: 12
[2022-10-06 16:38:53.606717] (error)   [fm]	 Unable to find a source for this conventional channel! Channel not added: 4.765000e+08 Talkgroup: 12
```

I propose a new top level control such as "perish_in_flames=true" that treats "warnings" like these as fatal.  I know it's been tested before and people complained, but adding it as a config flag would save "normal people" a lot of pain.  I would strongly suggest these being fatal makes a better default user experience as well.

faultywarrior, 2022-10-08:
I agree that this needs to be fixed.  It tripped me up as well when I once typoed the talkgroup file's filename (yes, even giving it a non-existent file will happily let it continue.)

A simple fix would be to add a check for if recordUnknown is set to true and the talkgroup file is invalid or doesn't exist, just error out the same way TR does if you have no sources which are able cover your control channels.

robotastic, 2022-10-10:
Good points - Having a malformed Channel File should definitely kill things... I don't think there is currently the expectation that it would keep working, even if there are some invalid entries. 

That talkgroup file checking has been pretty lack because it was originally just a nice-to-have that add some additional context. It is starting to become it a bit more critical in some configurations. both of these would be good error checks/fails to have.

 I will go make the switch and see what people say.

robotastic, 2022-11-27:
@ZeroChaos - Ok it took a little, but here is a branch that allows for optional strict processing of the .csv files: https://github.com/robotastic/trunk-recorder/tree/strict-parse

The `strictCsvParsing` config option at the top level should enable this. I still have to do some testing with it to make sure all of the different test cases work.

robotastic, 2024-12-14:
Addressed with  #811

closed 2024-12-14 by robotastic