TrunkRecorder/trunk-recorder #732: Feature Request: Ability To Blacklist TGs

faultywarrior, 2022-10-28:
I'd like to see a feature that allows you to use the talkgroups file to blacklist specific talkgroups instead of capturing them when used with the "record unknown talkgroups" option.  This way, you can still record dynamic TGs, but you can also ignore specific ones.  For example, I have a P25 Phase II system I capture which has a data beacon every 1 second on a specific TG; but they also use dynamic TGs for roaming users.  It'd be nice to allow capture of those without needing to put all 1000 possible TGs into a TG list just so that I can set 1 useless TG to a priority of 999.

tadscottsmith, 2022-10-28:
Can you not assign it -1?

Talkgroups assigned a priority of -1 will never be recorded, regardless of the number of available recorders.
https://github.com/robotastic/trunk-recorder/blob/master/docs/CONFIGURE.md#talkgroupsfile

faultywarrior, 2022-10-29:
This does appear to work.  I was under the impression that when record unknown talkgroups was enabled; the TG file was effectively ignored; but that does not appear to be the case.

I'm closing this as this does appear to work; although it probably should be documented a bit better.

closed 2022-10-29 by faultywarrior