TrunkRecorder/trunk-recorder #740: Talkgroups With A Priority Of -1 Still Get Recorded If Duplicated Across Systems

faultywarrior, 2022-12-04:
I capture 2 different P25 sites on the same trunked radio system (Same RFSS, Sysid, & WACN; but different Site ID's).  Each site is defined as a separate system in TR.  Often times because users for the same talkgroups are registered to both sites; I'll get duplicate calls recorded.  I've tried setting the duplicated talkgroups to a priority of -1 on the system that sees them less; but they still keep get recorded on both systems.  The -1 priority function appears to work if the talkgroup isn't in 2 systems; as a 3rd and totally seperate system I capture has a talkgroup at -1 which does not get recorded.  All settings are the same between the 3 defined systems in my config JSON sans the name, control channels, and talkgroup file name.

I've attached my config and talkgroup files.  The two system in question are Lewisville and Denton County.  Specifically, I'm trying to NOT capture calls that would be on the Lewisville site on the Denton County site.

[TR-Files.zip](https://github.com/robotastic/trunk-recorder/files/10147642/TR-Files.zip)

robotastic, 2022-12-04:
Ah - this is interesting! I just checked and Talkgroups are not associated with a system. So if you load a Talkgroup file up, it ends up applying to all of the systems. I should fix this. The weird thing is that I would think the opposite would happen and the -1 would apply to all of the systems and none of them would record it. Let me go walk through the code and make a few changes.

faultywarrior, 2022-12-04:
That looks to have done it!  The log now shows that the duplicate call wasn't recorded because the talkgroup has a priority of -1.

faultywarrior, 2022-12-05:
Alright, after running the latest build for ~24 hours I can confirm that it's both stable and resolves this issue.  Therefore, I'm closing this issue.  Thanks for the quick fix!

closed 2022-12-05 by faultywarrior