TrunkRecorder/trunk-recorder #743: More Stats for MQTT Statistics Plugin

s3m1s0n1c, 2022-12-09:
As discussed in Discord.. Would be awesome to get more stats send to MQTT
See below an example of what kind of stats that would be good to have.

* Calls Per TG - Eg /sonic/feeds/X/Y - X is the TG Number and Y is the calls count for that TG
* Calls per Unit - Eg /sonic/feeds/X/Y - X is the Unit Number and Y is the calls count for that Unit
* Call Error Rate - Eg /sonic/feeds/callrate/X - X is the Current Call Error Rate
* Call Rate Per System - /sonic/feeds/X/Y - X is the System Number and Y is the number of calls for that system

I'm sure others will be able to add more info on what stats they want to see

Thanks
Sonic

s3m1s0n1c, 2022-12-09:
Transmission per Agency would be awesome as well..

robotastic, 2024-12-14:
I think the MQTT plugin that @taclane put together has some additional features. Closing this, but it is a good reference if someone is able to make some updates