TrunkRecorder/trunk-recorder #748: recordings are missing end of transmissions

israelgoldstein, 2022-12-19:
some recording files is missing end of the transmissions 

4616-1671430493_851075000-call_18

1:14 AM
TGID 4616
Frequency: 851.0750

[12-19-2022_0112_00.log](https://github.com/robotastic/trunk-recorder/files/10256417/12-19-2022_0112_00.log)

israelgoldstein, 2022-12-19:
I've modified my config file, and removed all systems except the one where i noticed the issue

here is another example,

(log file should be clearer as its only for 1 system)

4001-1671432341_851075000-call_4

1:45 AM 
TGID 4001
Frequency: 851.0750

[12-19-2022_0143_00.log](https://github.com/robotastic/trunk-recorder/files/10256571/12-19-2022_0143_00.log)

israelgoldstein, 2022-12-19:
and 1 more log

22501-1671434237_851325000-call_17

2:17 AM
TGID 22501
Frequency: 851.3250

[12-19-2022_0207_00.log](https://github.com/robotastic/trunk-recorder/files/10256763/12-19-2022_0207_00.log)

tadscottsmith, 2022-12-19:
Looks like you've got some pretty serious signal issues or performance issues around that time. Any time you see parse errors, it's bad, and assuming this is P25 P1, you should be maintaining around 39-40 control channel messages per second. Can you try moving your antenna? A different antenna? Or playing with the gain settings to try and get a more consistent signal?
```
[2022-12-19 02:17:35.148452] (error)   P25 Parse error, s:  s0:   s1: t shift: 0 nac: 116 type: 65535 Len: 0
[2022-12-19 02:17:36.209048] (error)   P25 Parse error, s:  s0:   s1: t shift: 0 nac: 116 type: 65535 Len: 0
```

```
[2022-12-19 02:17:34.008429] (error)   [OrangeCoRNet]	 Control Channel Message Decode Rate: 6.33333/sec, count:  19
[2022-12-19 02:17:37.007260] (error)   [OrangeCoRNet]	 Control Channel Message Decode Rate: 4.66667/sec, count:  14
```

closed 2024-12-14 by robotastic