TrunkRecorder/trunk-recorder #752: Audio sounding "digitized" or "robotic" on new version?

bnelson333, 2022-12-23:
I tried poking around to see if I could find a post on this and didn't see anything, apologies if it's a duplicate. I upgraded to the latest version of trunk-recorder last week and ever since I have noticed a lot (but not all) of the audio clips saved sound "digitized" or "robotic". I thought it was my antennas or config or something but then I wondered if it was the version instead.

So this morning I rolled it back to the previous version (I was on 4.2.0) and sure enough, it seems to have solved the problem. So my question is - is this known? Is there something I need to do to fix for it in 4.4.0? Something in the config?

I believe this is the same dispatcher, so I have example audio files of the:

Before (4.2.0): https://sndup.net/j6kr/

and

After (4.4.0): https://sndup.net/wntn/

Thoughts?

tadscottsmith, 2022-12-23:
Version 4.2.0 appears to use the "newer" OP25 Vocoder. A number of people didn't like that sound, and at some point it was reverted back to the older version, which is now included in 4.4.0. The commit that reverted it appears to be https://github.com/robotastic/trunk-recorder/commit/a9e66dac238e14a48a2d5dbb9ee37a205f56c6f1. If you remove these changes, it would go back to the newer vocoder.

bnelson333, 2022-12-23:
So I could have 4.4.0 but with the vocoder from 4.2.0? Or do I have to go back to 4.2.0. completely? Also, I don't know how to install anything from git other than following the tutorial which just copies down the latest version. How do I do that with a non-current version?

bnelson333, 2022-12-24:
> Version 4.2.0 appears to use the "newer" OP25 Vocoder. A number of people didn't like that sound, and at some point it was reverted back to the older version, which is now included in 4.4.0. The commit that reverted it appears to be [a9e66da](https://github.com/robotastic/trunk-recorder/commit/a9e66dac238e14a48a2d5dbb9ee37a205f56c6f1). If you remove these changes, it would go back to the newer vocoder.

@robotastic would it be possible to request the ability to select the vocoder via config.json in a future release? I'd like to stay current but I find it difficult to hear what they're saying in the current vocoder (I like the one that was in 4.2.0).

robotastic, 2022-12-24:
I like - this sounds like a good idea and it should be pretty easy. I will add this to my backlog.

israelgoldstein, 2023-07-10:
Hi, was this done? im also facing issue with the audi sounding very digitized

closed 2023-07-10 by robotastic