TrunkRecorder/trunk-recorder #755: Not honoring CallTimeout

jhablitzel, 2022-12-27:
Monitoring a P25 PH1 system (Ohio MARCS).  All of my calls have the comment on the conclude line "Last Update: 3s".  I have my call timeout set to 1.   I'm not sure what the last update means, but it feels like my call timeout is not working.  I did verify on startup that the parameter gets loaded as 1 sec.
Another possible hint.  I upload to Broadcastify.  I believe they trim silence on those calls and the call times on Broadcastify are always 3-4 seconds less than what the log shows.  And my call skew on Broadcastify is always 4-5 seconds for some reason even though the program logs looks like the upload completes within milliseconds of the call conclude.

per tadscottsmith, "Not sure if it's intentional, or just some debugging that wasn't cleaned up, but config.call_timeout is currently being ignored in digital systems. https://github.com/robotastic/trunk-recorder/blob/77514919589a044064d6a947a9713d5e80d75fd3/trunk-recorder/main.cc#L822"

rosecitytransit, 2023-03-25:
If it'll help, you can set `verbosity=10` [here](https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/recorders/p25_recorder_decode.cc#L87) to see raw metadata; this should ideally be exposed as a configuration option and may be best to only do it with one recorder set as the data isn't tagged. The IMBEs and LDUs are actual voice frames, and TDUs are silence frames between and after transmissions

robotastic, 2024-12-14:
Closing this out - please reopen if it is still an issue

closed 2024-12-14 by robotastic