TrunkRecorder/trunk-recorder #760: Failing to Capture UIDs Again

Dewey3, 2022-12-31:
Greetings!  Since the issue in https://github.com/robotastic/trunk-recorder/issues/674 was fixed, I have purposely not updated since the beginning of September.  In order to not fall too far behind, I just updated to 4.5.0 last week.  I must unfortunately report that this latest version has the return of the problem where the UIDs as displayed by Rdio-Scanner are failing to change during "rapid-fire" transmissions.

Dewey3, 2023-02-13:
This is just a brief bump of the topic as the latest version, `commit 6fcc7f67cf16ca1af949d2b8326fa99299ad24e5 (master)` (2-7-2023) is still failing to capture the UIDs for the quick back-to-back transmissions.  The last known working copy for me was the `commit 10a276385124ef48e7ad6a3493236082c360ec0e (master)` (9-15-2022) version.  Is there anything I can do to assist in re-fixing this?

rosecitytransit, 2023-03-25:
If it'll help, you can set `verbosity=10` [here](https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/recorders/p25_recorder_decode.cc#L87) to see raw P25 voice channel metadata which should include the src ID in use; it may be best to only do it with one recorder set as the data isn't tagged.

Dewey3, 2023-03-27:
Thanks.  Not sure if I quite understand what will happen here.  I had the UID problem before and it was fixed back in August or September with the help of @tadscottsmith but newer releases seem to have broken it again.  Unfortunately, I don't know where the break is occurring, but I do know that I don't see the same problem when decoding with Unitrunker or monitoring on the BCDx35HP or SDS series scanners.

rosecitytransit, 2023-03-27:
Changing the setting and rebuilding will turn on a bunch of debug messages that show exactly what is being received in a voice call. Voice calls contain not only the actual voice but meta data like the talk group, unit ID and emergency/priority/etc flags about the call.

If you really wanted, I could possibly log in to your machine (or go over things together via Skype or whatever) and see what's happening.

Dewey3, 2023-03-27:
That is how @tadscottsmith was able to help me out, I gave him an account to remote in and work with it how he needed.  Are your saying the same, which is no problem, but it may be the weekend before I can setup the remote access?

rosecitytransit, 2023-03-27:
That's fine, just let me know when you have time

tadscottsmith, 2023-03-27:
That won't actually fix anything, it'll just spit out a bunch of logs.

@Dewey3 Just to confirm, you're seeing this issue again in v4.5.0 or later?

rosecitytransit, 2023-03-27:
Correct that it won't change anything, but it'll help identify what's
happening

robotastic, 2024-12-15:
Closing this and tracking in #836

closed 2024-12-15 by robotastic