TrunkRecorder/trunk-recorder #775: Too many control channels?

jtetterton2, 2023-02-13:
I have been monitoring a P25P2 system for quite some time now.. seems they have added some control channels. It's a Harris system so any freq can be the control channel. Initially there were 6 control channels and now there are 9. When launching trunk-recorder it cycles through the first 7 then exits.

Initial config that worked fine whenever an alternative control channel was being used:

        `"control_channels": [851437500.0,851975000.0,852312500.0,852462500.0,852887500.0,853887500.0],`

New config where it ignores the last 3 control channels then exits:

        `"control_channels": [851437500.0,851975000.0,852312500.0,852462500.0,852887500.0,853887500.0,857737500.0,858737500.0,859737500.0],`

The documentation at https://github.com/robotastic/trunk-recorder/blob/master/docs/CONFIGURE.md#system-object doesn't mention any cap on the number of control channels.

It seems to recognize all the control channels...

```
[2023-02-12 23:00:40.274975] (info)   Control Channels:
[2023-02-12 23:00:40.275032] (info)     851.437500 MHz
[2023-02-12 23:00:40.275151] (info)     851.975000 MHz
[2023-02-12 23:00:40.275214] (info)     852.312500 MHz
[2023-02-12 23:00:40.275273] (info)     852.462500 MHz
[2023-02-12 23:00:40.275331] (info)     852.887500 MHz
[2023-02-12 23:00:40.275390] (info)     853.887500 MHz
[2023-02-12 23:00:40.275448] (info)     857.737500 MHz
[2023-02-12 23:00:40.275506] (info)     858.737500 MHz
[2023-02-12 23:00:40.275563] (info)     859.737500 MHz
```

But only tries the first 7.

```[2023-02-12 23:01:15.806304] (info)   [albp25]  Started with Control Channel: 851.437500 MHz
[2023-02-12 23:01:15.807467] (info)      P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 10000000
[2023-02-12 23:01:15.819665] (info)      P25 Trunking ARB - Initial Rate: 10000000 Resampled Rate: 25000 Initial Decimation: 100 System Rate: 24000 ARB Rate: 0.96
[2023-02-12 23:01:16.871272] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:17.870928] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:18.002521] (error)   [albp25] Retuning to Control Channel: 851.975000 MHz
[2023-02-12 23:01:18.003087] (info)      - System Source 0 - Min Freq: 850.637500 MHz Max Freq: 860.537500 MHz
[2023-02-12 23:01:18.003797] (error)   [albp25]  Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2023-02-12 23:01:18.922558] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:19.953188] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:21.003270] (error)   [albp25] Retuning to Control Channel: 852.312500 MHz
[2023-02-12 23:01:21.003520] (info)      - System Source 0 - Min Freq: 850.637500 MHz Max Freq: 860.537500 MHz
[2023-02-12 23:01:21.003804] (error)   [albp25]  Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2023-02-12 23:01:21.003895] (error)   P25 Parse error, s:  s0:  s1: n shift: 0 nac: 110 type: 65535 Len: 0
[2023-02-12 23:01:22.043488] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:23.082904] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:24.011853] (error)   [albp25] Retuning to Control Channel: 852.462500 MHz
[2023-02-12 23:01:24.012140] (info)      - System Source 0 - Min Freq: 850.637500 MHz Max Freq: 860.537500 MHz
[2023-02-12 23:01:24.012513] (error)   [albp25]  Control Channel Message Decode Rate: 1/sec, count:  3
[2023-02-12 23:01:24.133863] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:25.173938] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:26.213887] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:27.011711] (error)   [albp25] Retuning to Control Channel: 852.887500 MHz
[2023-02-12 23:01:27.012165] (info)      - System Source 0 - Min Freq: 850.637500 MHz Max Freq: 860.537500 MHz
[2023-02-12 23:01:27.012514] (error)   [albp25]  Control Channel Message Decode Rate: 1/sec, count:  3
[2023-02-12 23:01:27.255142] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:28.294917] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:29.335085] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:30.011510] (error)   [albp25] Retuning to Control Channel: 853.887500 MHz
[2023-02-12 23:01:30.011791] (info)      - System Source 0 - Min Freq: 850.637500 MHz Max Freq: 860.537500 MHz
[2023-02-12 23:01:30.012061] (error)   [albp25]  Control Channel Message Decode Rate: 1/sec, count:  3
[2023-02-12 23:01:30.385548] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:31.425792] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:32.475536] (error)   P25 Parse error, s:  s0:  s1: ▒ shift: 0 nac: 246 type: 65535 Len: 0
[2023-02-12 23:01:33.011162] (error)   [albp25] Retuning to Control Channel: 857.737500 MHz
[2023-02-12 23:01:33.011444] (info)      - System Source 0 - Min Freq: 850.637500 MHz Max Freq: 860.537500 MHz
[2023-02-12 23:01:33.011755] (error)   [albp25]  Control Channel Message Decode Rate: 1/sec, count:  3
[2023-02-12 23:01:33.011890] (info)   Caught Exit Signal...
[2023-02-12 23:01:33.011961] (info)   Cleaning up & Exiting...
```

Version is 4.4.1.

tadscottsmith, 2023-02-14:
Disregard, I see your source should cover all of them.

```System Source 0 - Min Freq: 850.637500 MHz Max Freq: 860.537500 MHz```

jtetterton2, 2023-02-14:
Here is my config

```
{
    "ver": 2,
    "sources": [{
        "center": 855587500.0,
        "rate": 10000000.0,
        "error": 0,
        "gain": 25,
        "digitalRecorders": 16,
        "analogRecorders": 6,
        "driver": "osmosdr",
        "device": "airspy"
    }],
    "systems": [{
        "type": "p25",
        "control_channels": [851437500.0,851975000.0,852312500.0,852462500.0,852887500.0,853887500.0,857737500.0,858737500.0,859737500.0],
        "squelch": "-160",
        "talkgroupDisplayFormat": "tag_id",
        "shortName": "albp25",
        "talkgroupsFile": "/opt/recorder/talkgroupsAlbP25_all.csv",
        "modulation": "qpsk",
        "uploadScript": "./encode_upload_albp25.py",
        "broadcastifyApiKey": "REDACTED",
        "broadcastifySystemId": "REDACTED",
        "audioArchive": "true",
        "recordUnknown": "true",
        "hideUnknownTalkgroups": false
        }],

    "captureDir": "/opt/recorder/archive/",
    "frequencyFormat": "mhz",
    "controlWarnRate": 10,
    "controlRetuneLimit": 5,
    "callTimeout": 3,
    "broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
    "uploadServer": "https://api.openmhz.com",
    "logFile": "true",
    "logLevel": "info"
}

taclane, 2023-02-14:
> `    "controlRetuneLimit": 5,`

Ah.  Makes sense now.  You have a greater number of control channels than the retune limit. Trunk-recorder will attempt to retune the control channel that many times before gracefully exiting.  If your active control channel may be near the end of the list, it will never make it that far.  

Consider upping `controlRetuneLimit` to at least as many channels as you have, or omit the setting entirely in order to let trunk-recorder cycle through the list as long as it takes to find the control. If you find a frequency tends to be be a designated control, it can also help to put it at the beginning of the list to reduce the amount of time it will search for something that's active.

jtetterton2, 2023-02-14:
I totally missed that. I’ll up the limit and try again. This system rarely
switches control channels so this is the first time I’ve had this issue in
the last 7 months.

bctrainers, 2023-03-04:
> I totally missed that. I’ll up the limit and try again. This system rarely switches control channels so this is the first time I’ve had this issue in the last 7 months.

Since you have nine control channels on a single system, putting the `controlRetuneLimit` to three times the CC count would be wise. In your case, the `5` would be `27`. Allowing it to cycle through the entire list three times. TrunkRecorder is a bit slow on retune timeouts, unlike a Uniden, whistler, or Unication-based hardware where it is incredibly rapid. 

For systems with a smaller amount of CC/ACCs, talking three total CC/ACCs per system, it would be ideal for pushing it to 45, or about 15 attempts of frequency cycling. 

For whatever it's worth, and my 2c, your system has a stupid amount of CC/ACCs. There's legitimately no need for a P25 system to have _that many_ CCs. Legitimately wasting spectrum that would be better off used as additional data channels or voice channels. :/

closed 2024-12-14 by robotastic