TrunkRecorder/trunk-recorder #778: Question about RTL-SDR v5 SDR bandwidth

smokedsalmonbagel, 2023-02-18:
I am using several RTL-SDR v5 SDRs.  The device spec is 3.2 mhz of bandwidth.  When I try to use a rate much above `2000000` this in my config.json I get lots of noise.  Even strong signals are unintelligible (all conventional).  What is the limiting factor here?   Getting an extra 1mhz bandwidth would be great.

kd4e, 2023-02-18:
I switched to the nesdr smart sticks - they dissipate heat better & have 
been rock solid.

robotastic, 2023-02-19:
I have a fan pointed at my Nesdrs - I am not sure if it changes the noise floor, but it I haven't had a stick die, and they used too

taclane, 2023-02-20:
You may not reliably get more than 2.4 MHz out of one of those sticks (or a NESDR either).  Experiences can vary, but when trying to receive _and decode_ data, 2.56 MHz likely works fine, but 2.88 MHz starts pushing it.  Going even higher is fine when you're just _viewing_ the spectrum plot on GQRX, but that's because the samples that are getting lost aren't being decoded.

These just aren't purpose-built SDRs, it's a limitation of the RTL2832U DVB-T chip these devices are using.  Even though they are USB 2 devices, the test mode that is exploited for hobbyist SDR does not run at those speeds.  For both brands, the max advertised "stable" bandwidth is 2.4 MHz; the success of anything higher can depend on things like your USB host controller or cables.

As far as picking sample rates, try and stick to the classics:
```
240000, 300000, 960000, 1152000, 1200000, 1440000, 1600000,
1800000, 1920000, 2400000, 2560000, 2880000, 3200000,
```
While you can technically pick almost any sample rate, some (like the above examples) are much easier for trunk-recorder to break down into smaller channels without making the CPU work too hard.

smokedsalmonbagel, 2023-02-20:
Thank you @taclane - This is great to know.  I am trying to cover a system where there are 8 channels, each about 1 mhz apart.  I have a hackrf - but the NESDRs seem to be better (more stable and sensitive.)

smokedsalmonbagel, 2023-03-07:
Wanted to follow up before I close this.  I was able to use a sample rate of 2560000 maximum on 2 NESDR V5s.  As taclane mentioned using the suggested standard samples rates allowed me an extra 500khz sample rate or so.

closed 2023-03-07 by smokedsalmonbagel

rosecitytransit, 2023-04-17:
For the record, trunk-recorder cuts off some of the bandwidth in https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/source.cc#L419

The actual usable bandwidth that corresponds with the "classics" settings:
240000 192000
300000 250000
960000 896000
1152000 1088000
1200000 1100000
1440000 1344000
1600000 1536000
1800000 1700000
1920000 1792000
2400000 2300000
2560000 2432000
2880000 2816000
3200000 3072000