TrunkRecorder/trunk-recorder #781: use scan-build for testing

ZeroChaos-, 2023-02-24:
I don't know the "right" way to do this, but some kind folks in the #gentoo-dev-help channel on libera irc helped me discover scan-build

Min test case for use
```
scan-build cmake ..
scan-build make -j$(nproc)
```
This will build the code while using a static code analysis tool to report errors.  It seems to successfully find bugs like the ones found at runtime with https://github.com/robotastic/trunk-recorder/issues/780

It also makes a cool html report at the end.  I don't know how to do github actions well, but this would be an excellent way to add some automated testing.  This might be a good start to work on that https://github.com/marketplace/actions/run-cmake

ZeroChaos-, 2023-02-24:
Example output
```
zero@theprophet trunk-recorder % mkdir build                                                                                                                 (git)-[master] 
zero@theprophet trunk-recorder % cd build                                                                                                                    (git)-[master] 
zero@theprophet build % CXXFLAGS="-D_GLIBCXX_ASSERTIONS" scan-build cmake ..                                                                                 (git)-[master] 
scan-build: Using '/usr/lib/llvm/15/bin/clang-15' for static analysis
-- The CXX compiler identification is GNU 12.2.1
-- The C compiler identification is GNU 12.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/lib/llvm/15/libexec/c++-analyzer - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/lib/llvm/15/libexec/ccc-analyzer - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.8.1") 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'gmp'
--   Found gmp, version 6.2.1
-- Found GMP: /usr/lib64/libgmpxx.so  
-- Using GMP.
-- Found MPLIB: /usr/lib64/libgmpxx.so  
-- Found Boost: /usr/lib64/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.81.0") found components: date_time program_options system regex thread 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3.10
-- Found PythonInterp: /usr/bin/python3.10 (found version "3.10.9") 
-- Found PythonLibs: /usr/lib64/libpython3.10.so (found suitable exact version "3.10.9") 
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
-- Using GMP.
-- GnuRadio Version: 200709
-- Checking for module 'gnuradio-uhd'
--   Found gnuradio-uhd, version 3.10.5
-- Found gnuradio-uhd: /usr/include, /usr/lib64/libgnuradio-uhd.so
-- Found GnuradioUHD: /usr/lib64/libgnuradio-uhd.so  
-- Checking for module 'gnuradio-osmosdr'
--   Package 'gnuradio-osmosdr', required by 'virtual:world', not found
-- Found gnuradio-osmosdr: /usr/include, /usr/lib64/libgnuradio-osmosdr.so
-- Found GnuradioOsmosdr: /usr/lib64/libgnuradio-osmosdr.so  
-- Checking for module 'libhackrf'
--   Found libhackrf, version 0.8
-- Found LibHackRF: /usr/lib64/libhackrf.so  
-- Checking for module 'uhd'
--   Found uhd, version 4.3.0.0
-- Found LibUHD: /usr/lib64/libuhd.so  
-- Found OpenSSL: /usr/lib64/libcrypto.so (found version "1.1.1t")  
-- Found CURL: /usr/lib64/libcurl.so (found version "7.87.0")  
-- Configuring Boost C++ Libraries...
-- Found Boost: /usr/lib64/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: thread system program_options filesystem log_setup log regex random 
-- Gnuradio Version is: 3.10.5.1
-- AMBE Tone Synthesis Enabled
-- All libraries:
-- Configuring done
-- Generating done
-- Build files have been written to: /home/zero/development/trunk-recorder/build
scan-build: Analysis run complete.
scan-build: Removing directory '/tmp/scan-build-2023-02-24-161256-12601-1' because it contains no reports.
scan-build: No bugs found.
zero@theprophet build % scan-build make -j12                                                                                                                 (git)-[master] 
scan-build: Using '/usr/lib/llvm/15/bin/clang-15' for static analysis
[  0%] Checking the git repository for changes...
[  2%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/formatter.cc.o
[  2%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/call_impl.cc.o
[  3%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/call_conventional.cc.o
[... 1287 lines cut ...]
  ^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/zero/development/trunk-recorder/trunk-recorder/gr_blocks/wavfile_gr3.8.cc:112:3: warning: Value stored to 'fresult' is never read [deadcode.DeadStores]
  fresult = fread(&bits_per_sample, 1, 2, fp);
  ^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/zero/development/trunk-recorder/trunk-recorder/gr_blocks/wavfile_gr3.8.cc:155:3: warning: Value stored to 'fresult' is never read [deadcode.DeadStores]
  fresult = fread(&chunk_size, 1, 4, fp);
  ^         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
9 warnings generated.
[ 99%] Building CXX object CMakeFiles/trunk-recorder.dir/trunk-recorder/gr_blocks/rms_agc.cc.o
/home/zero/development/trunk-recorder/trunk-recorder/plugin_manager/plugin_manager.cc:69:7: warning: Value stored to 'ret' is never read [deadcode.DeadStores]
      ret = 0;
      ^     ~
1 warning generated.
/home/zero/development/trunk-recorder/trunk-recorder/call_concluder/call_concluder.cc:194:7: warning: Value stored to 'result' is never read [deadcode.DeadStores]
      result = system(shell_command_string.c_str());
      ^        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 warning generated.
[100%] Linking CXX executable trunk-recorder
[100%] Built target trunk-recorder
[100%] Linking CXX shared module ../../libstat_socket.so
[100%] Built target stat_socket
scan-build: Analysis run complete.
scan-build: 52 bugs found.
scan-build: Run 'scan-view /tmp/scan-build-2023-02-24-161307-12840-1' to examine bug reports.
```

ZeroChaos-, 2023-02-24:
[image]

robotastic, 2023-04-04:
This is pretty cool - you can install the scan-build tool in Ubuntu with `clang-tools` package: https://packages.ubuntu.com/kinetic/clang-tools