TrunkRecorder/trunk-recorder #787: Non control channels in config.json

smokedsalmonbagel, 2023-03-07:
Hi I looked over the documentation but could not find the answer:
I have a p25 system with the channels:
```
852.325
855.8125
856.4875c
857.4875c
858.4875c
859.7625c
```
Control channels are marked with a 'c'.   Do the non control channels need to be specified?  If so, where? If they don't do they need to be within my covered source range?

natecarlson, 2023-03-07:
No, you don't specify the non-control channels in the configuration. trunk-recorder will lock on to the active control channel and use that. The control messages then specifies what frequencies given transmissions are on.

Your sources do need to cover all the possible frequencies; there will still be transmissions on the frequencies that aren't control channels.

smokedsalmonbagel, 2023-03-07:
Perfect - thanks for the quick reply.

closed 2023-03-07 by smokedsalmonbagel