TrunkRecorder/trunk-recorder #789: Deduplicate TGs on simulcasted sites with multisite

smokedsalmonbagel, 2023-03-18:
I monitor several sites.  Two are simulcast sites with many overlapping talk groups.  Some TGs are rebroadcast on these adjacent sites.  I tried multisite options but it seems like it is not working - I still see the same recording from both systems.  Is there a work around / feature for this?  I am not sure this is the use case for the multisite option - both sites do have the same WACN.  Here is my config.

```
{
    "ver": 2,
    "captureDir":"/media",
    "sources": [
        
        {
            "center":771700000,
            "rate": 2400000,
            "error": 0,
            "gain": 30,
            "driver": "osmosdr",
            "device": "rtl=02,buflen=65536",
            "digitalRecorders": 4
        },
        {
            "center":852200000,
            "rate": 2400000,
            "error": 0,
            "gain": 30,
            "driver": "osmosdr",
            "device": "rtl=05,buflen=65536",
            "digitalRecorders": 4
        },
        {
            "center":854500000,
            "rate": 2400000,
            "error": 0,
            "gain": 30,
            "driver": "osmosdr",
            "device": "rtl=06,buflen=65536",
            "digitalRecorders": 4
        },
        {
            "center":856800000,
            "rate": 2400000,
            "error": 0,
            "gain": 30,
            "driver": "osmosdr",
            "device": "rtl=09,buflen=65536",
            "digitalRecorders": 4
        },
        {
            "center":859100000,
            "rate": 2400000,
            "error": 0,
            "gain": 30,
            "driver": "osmosdr",
            "device": "rtl=10,buflen=65536",
            "digitalRecorders": 4
        },
        {
            "center":773500000,
            "rate": 1200000,
            "error": 0,
            "gain": 30,
            "driver": "osmosdr",
            "device": "rtl=04,buflen=65536",
            "digitalRecorders": 4
        },
        {
            "center":769181250,
            "rate": 600000,
            "error": 0,
            "gain": 30,
            "driver": "osmosdr",
            "device": "rtl=03,buflen=65536",
            "digitalRecorders": 4
        }
    ],
    "systems": [
        {
            "control_channels": [773056250,773956250,852700000,853312500],
            "type": "p25",
            "shortName": "Troop_A",
            "multiSite" : true,
            "multiSiteSystemName":"CSP",
            "talkgroupsFile":"talkgroups.csv",
            "minTransmissionDuration":2,
            "modulation": "qpsk",
            "uploadScript":"/app/encode-upload.sh",
            "audioArchive" : false,
            "recordUnknown":true,
            "compressWav" : false,
            "decodeMDC" : true,
            "decodeFSync" : true,
            "decodeStar" : true,
            "decodeTPS" : true
        },
        {
            "control_channels": [852900000,852112500,852175000,851187500,851337500,851587500,851837500,851900000,853650000],
            "type": "p25",
            "shortName": "Waterbury",
            "minTransmissionDuration":2,
            "talkgroupsFile":"talkgroups_wby.csv",
            "modulation": "qpsk",
            "uploadScript":"/app/encode-upload.sh",
            "audioArchive" : false,
            "recordUnknown":true,
            "compressWav" : false,
            "decodeMDC" : true,
            "decodeFSync" : true,
            "decodeStar" : true,
            "decodeTPS" : true
        },
        {
            "control_channels": [859762500,856487500,857487500,858487500],
            "type": "p25",
            "multiSite" : true,
            "shortName": "Troop_L",
            "multiSiteSystemName":"CSP",
            "minTransmissionDuration":2,
            "talkgroupsFile":"talkgroups.csv",
            "modulation": "qpsk",
            "uploadScript":"/app/encode-upload.sh",
            "audioArchive" : false,
            "recordUnknown":true,
            "compressWav" : false,
            "decodeMDC" : true,
            "decodeFSync" : true,
            "decodeStar" : true,
            "decodeTPS" : true
        }
    ]
}
```

natecarlson, 2023-03-18:
Try removing the system name; I ran into the same thing and the author of the feature and I discovered that the code isn't behaving as expected there. The system name isn't needed for P25.

faultywarrior, 2023-03-18:
1 - CLMRN is a mess for overlaps since troopers can be damn near anywhere in the state and their radios wind up affiliated to whatever site they get a better RSSI to; so you're bound to get crossing; esp. on Troop L where they cross into western Hartford county a lot.  I had all kinds of fun trying to capture the old SmartZone system with site overlap.

2 - You can make each site a separate system and then use the TG list to blacklist stuff.  i.e. blacklist all of Troop L's TGs from Troop H's site, etc. I had the same issue and opened #732 where someone suggested this.  You If you leave recordunknown true, you can put only the TGs you want to blacklist from each site and set them to a priority of -1.   I do this for the systems I monitor down here in Texas and it works works quite well.

smokedsalmonbagel, 2023-03-18:
> Try removing the system name; I ran into the same thing and the author of the feature and I discovered that the code isn't behaving as expected there. The system name isn't needed for P25.
Thanks - I think I had the name removed on the first try, it sounded like it was needed for smartnet - but I will remove and see what the result is this evening.

@faultywarrior 
Good to know - for the most common TGs I have been using recordunknown false and removing the TGs from the talkgroups file.  I have a talkgroups file for each site with some of the most common repeated talkgroups removed.  But I see the benefit of the method describe in #732.
I have antennas at around 1150' near troop L so yes - duplicates from A, B, I and H.

I will keep working on this - it might be necessary to put together a post processing duplicate removal script.

tadscottsmith, 2023-03-18:
If it's this system, [CLMRN](https://www.radioreference.com/db/sid/7196), try removing `"multiSiteSystemName":"CSP"` from both, as suggested. If that doesn't work, could you grab some logs around the times the duplicate calls occur? I'm not sure if anyone has tested on P25 P2, and I could have missed something.

natecarlson, 2023-03-20:
OK, so there is nothing in the logs showing that multisystem is doing anything.

```
[2023-03-20 10:09:07.243375] (info)   [Troop_L] 20023C  TG:       5662  Freq: 856.487500 MHz    Starting P25 Recorder Num [12]  TDMA: true      Slot: 1 QPSK: true
[2023-03-20 10:09:07.324572] (info)   [Troop_A] 20024C  TG:       5662  Freq: 851.637500 MHz    Starting P25 Recorder Num [4]   TDMA: true      Slot: 0 QPSK: true
[2023-03-20 10:09:15.012841] (info)   [Troop_L] 20023C  TG:       5662  Freq: 856.487500 MHz    Concluding Recorded Call_impl - Last Update: 4s Call_impl Elapsed: 8
[2023-03-20 10:09:15.012898] (info)   [Troop_L] 20023C  TG:       5662  Freq: 856.487500 MHz    Stopping P25 Recorder Num [12]  TDMA: true      Slot: 1 Hz Error: 30
[2023-03-20 10:09:15.012924] (info)   [Troop_L] 20023C  TG:       5662  Freq: 856.487500 MHz    - Transmission src: 1871011 pos: 0 length: 2.72
[2023-03-20 10:09:15.013015] (info)   [Troop_A] 20024C  TG:       5662  Freq: 851.637500 MHz    Concluding Recorded Call_impl - Last Update: 4s Call_impl Elapsed: 8
[2023-03-20 10:09:15.013061] (info)   [Troop_A] 20024C  TG:       5662  Freq: 851.637500 MHz    Stopping P25 Recorder Num [4]   TDMA: true      Slot: 0 Hz Error: -562
[2023-03-20 10:09:15.013088] (info)   [Troop_A] 20024C  TG:       5662  Freq: 851.637500 MHz    - Transmission src: 1871011 pos: 0 length: 2.68
[2023-03-20 10:09:15.016376] (info)   [Troop_A] 20024C   Running upload script: /app/encode-upload.sh /media/Troop_A/2023/3/20/5662-1679306947_851637500-call_20024.wav /media/Troop_A/2023/3/20/5662-1679306947_851637500-call_20024.json /media/Troop_A/2023/3/20/5662-1679306947_851637500-call_20024.m4a
[2023-03-20 10:09:15.016774] (info)   [Troop_L] 20023C   Running upload script: /app/encode-upload.sh /media/Troop_L/2023/3/20/5662-1679306947_856487500-call_20023.wav /media/Troop_L/2023/3/20/5662-1679306947_856487500-call_20023.json /media/Troop_L/2023/3/20/5662-1679306947_856487500-call_20023.m4a
Encoding: /media/Troop_L/2023/3/20/5662-1679306947_856487500-call_20023.wav
Encoding: /media/Troop_A/2023/3/20/5662-1679306947_851637500-call_20024.wav
{'avgafp': 1000.0, 'started': 1679306947, 'flt': 1, 'nid': '3r', 'avgrf': None, 'ended': 1679306950, 'mode': None, 'eventdt': '2023-03-20 10:09:07', 'duration': 3, 'path': '/media/Troop_A/2023/3/20/5662-1679306947_851637500-call_20024.mp3', 'freq': 851637500, 'type': 'rxsaved-rtl', 'rid': '1r', 'sys_name': 'Troop_A', 'chname': 'ServPtl-Newingtn', 'trunk_data': {'freq': 851637500, 'start_time': 1679306947, 'stop_time': 1679306950, 'emergency': 0, 'encrypted': 0, 'call_length': 3, 'talkgroup': 5662, 'talkgroup_tag': 'ServPtl-Newingtn', 'talkgroup_description': 'Service Patrol Newington Operations', 'talkgroup_group_tag': 'Public Works', 'talkgroup_group': 'Public Works', 'audio_type': 'digital tdma', 'short_name': 'Troop_A', 'freqList': [{'freq': 851637500, 'time': 1679306947, 'pos': 0.0, 'len': 2.68, 'error_count': '0', 'spike_count': '0'}], 'srcList': [{'src': 1871011, 'time': 1679306947, 'pos': 0.0, 'emergency': 0, 'signal_system': '', 'tag': ''}]}}
{'avgafp': 1000.0, 'started': 1679306947, 'flt': 1, 'nid': '3r', 'avgrf': None, 'ended': 1679306950, 'mode': None, 'eventdt': '2023-03-20 10:09:07', 'duration': 3, 'path': '/media/Troop_L/2023/3/20/5662-1679306947_856487500-call_20023.mp3', 'freq': 856487500, 'type': 'rxsaved-rtl', 'rid': '1r', 'sys_name': 'Troop_L', 'chname': 'ServPtl-Newingtn', 'trunk_data': {'freq': 856487500, 'start_time': 1679306947, 'stop_time': 1679306950, 'emergency': 0, 'encrypted': 0, 'call_length': 3, 'talkgroup': 5662, 'talkgroup_tag': 'ServPtl-Newingtn', 'talkgroup_description': 'Service Patrol Newington Operations', 'talkgroup_group_tag': 'Public Works', 'talkgroup_group': 'Public Works', 'audio_type': 'digital tdma', 'short_name': 'Troop_L', 'freqList': [{'freq': 856487500, 'time': 1679306947, 'pos': 0.0, 'len': 2.72, 'error_count': '0', 'spike_count': '0'}], 'srcList': [{'src': 1871011, 'time': 1679306947, 'pos': 0.0, 'emergency': 0, 'signal_system': '', 'tag': ''}]}}
```

Those look like identical calls that should have theoretically been caught?

What did your config.json look like for this log?

tadscottsmith, 2023-03-20:
Can you confirm what version of the docker image you're using? I don't believe multiSite is in 4.5, so it would probably only be available in `edge` or `nightly`.

Additionally, in your `Troop_A` configuration you have `recordUnknown` set to both true and false. I don't think that's related, but you might want to clean that up.

smokedsalmonbagel, 2023-03-20:
OK - good to know.   I was using `latest` which was only at 4.4.2.  I am having issues with `nightly` getting an error about `channel` being an undefined node, but I have no node named `channel`.  It looks like `edge` is at 4.5 which starts up for me.  I can build the docker image from the most recent commit on the master branch.  Should I be looking in specific branch for the multisite functionality. Would like to help test / evaluate.

smokedsalmonbagel, 2023-04-09:
I am now seeing 
```
[2023-04-09 03:31:34.056813] (info)   [Troop_L] 2186C   TG:       8701  Freq: 855.812500 MHz    Starting P25 Recorder Num [12]  TDMA: true      Slot: 1 QPSK: true
[2023-04-09 03:31:34.319962] (info)   [Troop_A] 2187C   TG:       8701  Freq: 769.081250 MHz    Duplicate Grant. Original Call NAC: 929 Grant Message NAC: 928 Source: 0 Call: 2186C State: recording
```
So it looks like it was just the version I was running - thanks for this feature.  I will continue to test.

tadscottsmith, 2023-04-13:
Perfect! Can you close this issue and open a new one if anything new comes up?

closed 2024-12-14 by robotastic