TrunkRecorder/trunk-recorder #790: fdkaac: not found (on Debian 11)

bhall7, 2023-03-20:
Howdy! I'm getting the error `fdkaac: not found` when trying to run trunk-recorder. I'm running Debian 11 Bullseye, and when I attempt to `sudo apt-get install fdkaac` I just get "Unable to locate package fdkaac". I tried adding `non-free` to my `/etc/apt/sources.list`:

  `deb http://deb.debian.org/debian/ bullseye main non-free`

Then running `sudo apt-get update`, but no matter what, it still says "Unable to locate package fdkaac".

Any tips for getting this dependency installed on Debian 11 would be greatly appreciated.

Thanks!

gofaster, 2023-03-20:
Try adding the `contrib` package repo.

bhall7, 2023-03-21:
Thanks! I was able to install fdkaac after adding the `contrib` package repo. Thank you so much!

Although, after installing it, and trying to run trunk-recorder, it's getting hung up in the PLUGINS section:

```
-------------------------------------
PLUGINS
-------------------------------------
[2023-03-20 18:02:24.951174] (info)   Setting up plugin -  Name: openmhz_uploader	 Library file: libopenmhz_uploader.so
[2023-03-20 18:02:24.956825] (info)   Setting up plugin -  Name: broadcastify_uploader	 Library file: libbroadcastify_uploader.so
[2023-03-20 18:02:24.957344] (info)   Setting up plugin -  Name: unit_script	 Library file: libunit_script.so
[2023-03-20 18:02:24.957558] (info)   Setting up plugin -  Name: stat_socket	 Library file: libstat_socket.so
[2023-03-20 18:02:24.958508] (info)   No plugins configured
[2023-03-20 18:02:24.958528] (info)   **[getting hung up here]**
```
Is there a debug command-line option?

bhall7, 2023-03-21:
Weird...a reboot fixed it, and it's working fine now. Thanks! :-)

closed 2023-03-21 by bhall7