TrunkRecorder/trunk-recorder #792: Not tuning control channels on multiple sources.

jessm33, 2023-03-29:
One system I monitor recent switched control channels and trunk-recorder wouldn't switch to the new control channel, it kept cycling thru all 4 control channels.   Restarting trunk-recorder did not fix the issue.  Reordering the control channels in the config so the current control channel was first in the config fixed the issue.  Further testing shows it will tune the control channel, even if it is not the first one listed in config file, if it doesn't need to switch sources, before getting to the active control channel.

My config file is attached:
[config.json.txt](https://github.com/robotastic/trunk-recorder/files/11095390/config.json.txt)

jessm33, 2023-03-29:
I am running 4.5.0, there wasn't anything useful in the logs(some p25 parse errors, and control rate message from not being able to tune the control channel).  I've not seen any crashes when this happens, last time this happened it was stuck in the cycling thru control channels but kept running for a few hours before I could update the config to work around this issue.

I have found in my case this retune issue is caused by having different sample rates on the 2 sources with possible control channels.  Changing the sample rates for these 2 sources to same value allows trunk-recorder to retune across the sources.

taclane, 2023-03-29:
> I have found in my case this retune issue is caused by having different sample rates on the 2 sources with possible control channels. Changing the sample rates for these 2 sources to same value allows trunk-recorder to retune across the sources.

Out of curiosity, can you share what the control channel freq is, and the order they were in that worked, and *didn't work?*

Might be onto something with the sample rates; it's possible something isn't getting reset when it switches to a new source that sometimes, but not always, causes an issue.  I'd suspect that it probably isn't just the sample rate itself, but but some downstream calculation like decimation or prefilter settings.

It should be pretty to replicate this on other systems by using mismatched sample rates and throwing a couple dummy freqs into the control channel list.

jessm33, 2023-03-30:
The system has these 4 control channels, with my sdr setup the first 2 are on one source, and the last 2 are on another source.
856262500,
857262500,
858262500,
859262500

With 856262500, as the active control channel the original order works fine.

With this order trunk-recorder and different sample rates on the 2 sources, trunk recorder will loop continuously and  never lock on to a source.

858262500,
859262500,
856262500,
857262500

With the above order, and the same sample rates on both sources, trunk-recorder will correctly try the first 2 freqs on one source then move on the next source and lock on to the control channel.

robotastic, 2023-04-04:
OMG - that is non-trivial... It honestly maybe better to just kill the system block and create a new one. 

so... gave that a try! It compiles, but it may dump memory at an alarming rate. I am not sure how well everything gets cleaned up each time a new trunking decoder block is created.

Anyhow - can you give the `cc-retune` branch a try?

gonzoblue, 2023-05-07:
Looks like I had a similar issue. Even with only a single CC or the primary CC as the first in the json it wouldn't lock on. I tried the cc-retune branch and it immediately started working again.

robotastic, 2023-05-10:
OK - Seems like it is a good idea to merge this branch in!

closed 2023-05-10 by robotastic