TrunkRecorder/trunk-recorder #795: [RFC] Zig based build system.

Dygear, 2023-04-11:
As I had posted in the [Discord Server]() I think it might be worth moving over to a Zig based build system for Trunk-Recorder. This is mostly to make the build times faster, but also allow for cross-compile to different systems are architectures from the same host. You could for example use a linux-x86_64 to build for macos-aarch64 or even linux-aarch64. So large servers could built multiple versions of Trunk-Recorder for release quite easily.

This whole thing came about because I saw this video on YouTube called [Intro to the Zig Programming Language • Andrew Kelley • GOTO 2022](https://youtube.com/embed/YXrb-DqsBNU?start=464&stop=831) where Andrew Kelley shows how to use the features where Zig is a built system for C/C++. This allows for easy integration with Zig plugins later on. (Keep in mind, that I'm mostly a Rust programmer, so I have no real skin in this game, but I do think that using the Zig build system would be beneficial to the project overall.)

So if there are no objections, and if @robotastic would be willing to merge the work, if I were to actually do it, I would be willing to spend the time and make it happen.

robotastic, 2024-12-14:
I am late to the party ... but I would be open to other types of build systems. The trick would also converting over the OP25 build system, since TR pulls in a lot of that code. I am going to close for now, but feel free to re-open if you want to try and take this on.

closed 2024-12-14 by robotastic

Dygear, 2024-12-17:
This one went the way of the dodo as Zig started to move away from its “We can compile anything” system. They may have reversed that, but it’s not worth the headache at this point if they change their minds again.