TrunkRecorder/trunk-recorder #813: create recording mode option

rosecitytransit, 2023-05-12:
Mentioned in 791 

I've thought the recording mode should be at least a 3 way option.

Upon termination flag (transmission end):
- call is always closed, putting each transmission in a separate call (future transmissions are started based upon their grant messages by ignoring check for existing call)
- (current mode) file is closed but call is kept open
- file is kept open and closed on call end

To do conversion mode only involves:
1. not closing the file in `end_transmission()` here https://github.com/robotastic/trunk-recorder/blob/e33e441278ae130a6884d464a5d75b97655ca5fd/trunk-recorder/gr_blocks/transmission_sink.cc#L224
2. adding `close_wav()` to `stop_recording()` https://github.com/robotastic/trunk-recorder/blob/e33e441278ae130a6884d464a5d75b97655ca5fd/trunk-recorder/gr_blocks/transmission_sink.cc#L257 since `end_transmission()` will no longer do it
3. having a separate sample counter so that `close_wav()` creates a correct header
4. Ignoring or removing the error messages about the file not closing/being open
5. modifying call concluder to handle the entire call being in the first transmission file

_Originally posted by @rosecitytransit in https://github.com/robotastic/trunk-recorder/issues/791#issuecomment-1537242754_

robotastic, 2024-12-14:
Keeping the current approach - it is a good balance of keeping the number of files reasonable and not having to move file pointers around too. Happy to reconsider if there is a strong desire for this

robotastic, 2024-12-15:
Actually... maybe we should enable some support for turning off conversation mode.... I will leave this open

mikecomer-beakon, 2025-05-08:
Have you thought about doing this for talkgroups too not just system level?

robotastic, 2025-05-10:
Ok - I went through the code a little. I forgot  that with the new method of handling transmissions, everything is already chopped up into single transmissions and then merged back together at the end. To enable this we just need to not merge things together, and then just batch run the plugins on each file.

I will look to set it up so it can either be configured at the system level or by talkgroup

robotastic, 2025-05-13:
Ok - I just kicked off a branch - it compiles but has not been tested yet. It only has support for a system level setting currently:
https://github.com/TrunkRecorder/trunk-recorder/tree/dev/conversation

robotastic, 2025-05-23:
Ok - if folks can give this a try, that would great. @mikecomer-beakon I added in support for defining this at the talkgroup level and the system level. You can turn it on/off at the System Level, by adding conversationMode=true/false to a system in the config file. You can also override that at the Talkgroup level. Just add a "Conversation Mode" column to the talkgroups csv and define it for the TGs you are interested in.

rosecitytransit, 2026-05-02:
Getting back into t-r setting up an additional recording station and I think this is something that still could be implemented.