TrunkRecorder/trunk-recorder #815: JSON5 parser for config.json

Dygear, 2023-05-18:
Would it be possible to move over to the more forgiving [JSON5 spec](https://spec.json5.org/) for the `config.json` file? The reason for this, is that from time to time I think myself and many others add a trailing `,` comma to our objects or arrays. This is allowed in many programming languages as a quality of life improvement. I'd like to see the same thing in Trunk-Recorder's JSON parser that I feel is a little too strict.

This seems like a good choice for the library to use for JSON5 parsing.

https://github.com/nlohmann/json

robotastic, 2023-05-18:
I like that idea! I have had that problem too. This could be a good opportunity to move the Config file processing out of the main.cc file. I was just using the Boost JSON parsing because it was built in.

Looks like we could also enable trailing commas on Boost:
https://www.boost.org/doc/libs/1_76_0/libs/json/doc/html/json/ref/boost__json__parse_options/allow_trailing_commas.html
and comments:
https://www.boost.org/doc/libs/1_76_0/libs/json/doc/html/json/ref/boost__json__parse_options/allow_comments.html

Are there other JSON5 things that would be nice, or would that cover most of it?

Dygear, 2023-05-19:
I would like to be able to use C style comments `//` and `/* */` blocks. This is good because in Discord I can help someone with their config file, or even in the readme.md on the main page here, we can put the documentation right along side the examples so they don't have to go to far.

robotastic, 2023-05-19:
Nevermind... the read_json() function for reading into a Property Tree does not support those options...
https://www.boost.org/doc/libs/1_63_0/doc/html/boost/property_tree/json_parser/read_json_idp909550496.html

Dygear, 2023-05-19:
https://www.boost.org/doc/libs/1_76_0/libs/json/doc/html/json/ref/boost__json__parse_options/allow_comments.html

So close too!

So, I'd have to write a JSON5 to BOOST PTree bridge. Boo.

robotastic, 2024-12-14:
We did get an improved JSON parser in v5... it is not perfect but a lot better. Please reopen if a better JSON lib pops up

closed 2024-12-14 by robotastic

Dygear, 2024-12-17:
Oh, sorry didn’t realize this was still open. Thanks for closing.