TrunkRecorder/trunk-recorder #818: make fails at [ 74%] Building CXX object CMakeFiles/trunk_recorder_library.dir/lib/gr-latency/latency_probe.cc.o

bnelson333, 2023-05-27:
Ubuntu server 20.04, never had a problem building trunk-recorder before, now i'm getting this error. Thought maybe it was gcc-9, so i tried it with gcc-10 and still get the same error. Any ideas why it's failing? 

Error:
[ 53%] Building CXX object CMakeFiles/trunk_recorder_library.dir/lib/gr-latency/latency_probe.cc.o
/home/user/trunk-recorder/lib/gr-latency/latency_probe.cc: In static member function ‘static gr::gr_latency::latency_probe::sptr gr::gr_latency::latency_probe::make(int, std::vector<std::__cxx11::basic_string<char> >)’:
/home/user/trunk-recorder/lib/gr-latency/latency_probe.cc:37:42: error: could not convert ‘gnuradio::get_initial_sptr(T*) [with T = gr::gr_latency::latency_probe]()’ from ‘boost::shared_ptr<gr::gr_latency::latency_probe>’ to ‘gr::gr_latency::latency_probe::sptr’ {aka ‘std::shared_ptr<gr::gr_latency::latency_probe>’}
   37 |         return gnuradio::get_initial_sptr(new latency_probe (item_size, keys));
      |                ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                          |
      |                                          boost::shared_ptr<gr::gr_latency::latency_probe>
make[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:414: CMakeFiles/trunk_recorder_library.dir/lib/gr-latency/latency_probe.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:296: CMakeFiles/trunk_recorder_library.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

robotastic, 2023-05-28:
I just removed that file from the build... give it a try again, I think it should work now.

bnelson333, 2023-05-28:
Thanks that worked!

closed 2023-05-28 by bnelson333