TrunkRecorder/trunk-recorder #821: Add config switch for choosing full/half-rate p25 decoder

HerpMcDerperson, 2023-05-31:
Enough people, including myself, have experienced aural-molestation on account of the newer p25p1 failing to mute on non-audio packets. There should be a configuration switch for enabling one or the other rather than having to change which block is or isn't commented-out around https://github.com/robotastic/trunk-recorder/blob/master/lib/op25_repeater/lib/p25p1_fdma.cc#L723-L745

tadscottsmith, 2023-05-31:
Just to clarify, this is not related to full-rate/half-rate vocoders. That refers to the method of signal separation used by the P25 standard. This is more specifically to OP25's two different implementations of the vocoder, colloquially known as the "IMBE Decoder" and the "Software Decoder". 

- Full-Rate Vocoder (FDMA) - P25 P1 Standard.
- Half-Rate Vocoder (TDMA) - P25 P2 Standard.
- IMBE/Hardware Decoder - OP25's original, "fixed-point" implementation based on the P25 standards.
- Software Decoder - OP25's newer "software implementation" of the P25 standards with some additional P25 P2 capabilities (that I don't believe trunk-recorder is utilizing).

Just want to make sure we don't make an inaccurate config option based on the full/half-rate confusion.

HerpMcDerperson, 2023-05-31:
I was just parroting what I thought I understood about the discussion months ago. I now notice that the code clearly states that both blocks are related to full-rate decoding. I am pretty certain we both know what I am ACTUALLY talking about:
`// This is the Vocoder that OP25 currently uses.`
vs
`// This is the older, fullrate vocoder
// it was copied from p25p1_voice_decode.cc`
where, currently, the former is commented-out.
In order to wrangle some aural-assaulting chirps from my city's p25p1 conventional system, I needed to swap those. Others have run into it as well... I wonder if the particular sort of p25p1 systems we monitor share something in common.

robotastic, 2024-12-14:
a config option was added in to support choosing the vocoder

closed 2024-12-14 by robotastic