TrunkRecorder/trunk-recorder #826: t-r retuning to control channels on different system

rosecitytransit, 2023-06-10:
I'm using [a modified version of the new_call_mgmt branch](https://github.com/rosecitytransit/trunk-recorder/tree/trimet4.5), including the cc-retune fix, and things are working pretty good. But I've noticed twice now that recordings start getting saved into the wrong system.

I looked at the logs when it started this time and it seems that after trying the last defined control channel, trunk-recorder isn't going back to the first one but instead trying ones in a different system.

Control channels defined in config.json are as follows; `controlRetuneLimit` is not set:
```
...
"control_channels": [769856250, 770343750, 770093750, 771268750, 771693750, 771768750, 772068750, 772968750],
"shortName": "ms",
...
"control_channels": [772056250, 770081250, 770331250, 771206250, 771256250, 771481250, 771681250, 771756250, 772056250, 772356250, 773231250],
"shortName": "cc",
...
"control_channels": [773031250, 769456250, 770943750, 770443750, 773831250, 774568750, 773306250, 773568750, 774281250, 774831250],
"shortName": "osrppdxc",
...
"control_channels": [772606250, 771131250, 773131250, 85266250, 854462500, 770606250, 852100000, 852387500, 853637500, 856737500, 856887500, 857062500, 857462500, 857687500],
"shortName": "pdx2",
...
```
Log excerpt:
```
[2023-06-07 19:01:12.018369] (error)   [ms]      Control Channel Message Decode Rate: 2/sec, count:  7
[2023-06-07 19:01:15.022090] (error)   [ms] Retuning to Control Channel: 770.343750 MHz
[2023-06-07 19:01:15.022235] (error)   [ms]      Control Channel Message Decode Rate: 1/sec, count:  4
[2023-06-07 19:01:18.021302] (error)   [ms] Retuning to Control Channel: 770.093750 MHz
[2023-06-07 19:01:18.021419] (error)   [ms]      Control Channel Message Decode Rate: 1/sec, count:  3
[2023-06-07 19:01:21.015765] (error)   [ms] Retuning to Control Channel: 771.268750 MHz
[2023-06-07 19:01:22.568236] (error)   [ms]      Control Channel Message Decode Rate: 1/sec, count:  3
[2023-06-07 19:01:24.015650] (error)   [ms] Retuning to Control Channel: 771.693750 MHz
[2023-06-07 19:01:24.015909] (error)   [ms]      Control Channel Message Decode Rate: 0/sec, count:  2
[2023-06-07 19:01:27.022323] (error)   [ms] Retuning to Control Channel: 771.768750 MHz
[2023-06-07 19:01:27.022434] (error)   [ms]      Control Channel Message Decode Rate: 1/sec, count:  3
[2023-06-07 19:01:30.021362] (error)   [ms] Retuning to Control Channel: 772.068750 MHz
[2023-06-07 19:01:30.021489] (error)   [ms]      Control Channel Message Decode Rate: 1/sec, count:  3
[2023-06-07 19:01:33.015352] (error)   [ms] Retuning to Control Channel: 772.968750 MHz
[2023-06-07 19:01:33.015472] (error)   [ms]      Control Channel Message Decode Rate: 1/sec, count:  3
[2023-06-07 19:01:36.015707] (error)   [ms] Retuning to Control Channel: 774.281250 MHz
[2023-06-07 19:01:37.656044] (error)   [ms]      Control Channel Message Decode Rate: 1/sec, count:  3
[2023-06-07 19:01:39.013996] (error)   [ms] Retuning to Control Channel: 774.568750 MHz
[2023-06-07 19:01:39.014132] (error)   [ms]      Control Channel Message Decode Rate: 0/sec, count:  2
```
The first 7 retunes are as expected, but then trunk-recorder tries 2 channels on a completely different system (`osrppdxc`) and not even ones in a discernible order. The first one tried is parallel with where a next one on the `ms` system would be, but the next one seems random.  And trunk-recorder does indeed lock onto the `osrpppdxc` control channel and start recording calls from it into the `ms` system.

Note that I now have a Pi 3B+ that I was given up and running, and intend use that as a test bed if someone has some ideas to try

robotastic, 2023-06-10:
That is weird - I just made a change to the `master` branch to print out all of the Control Channels that were added for a system after it reads in the config file. Could you try running the this branch and seeing if the problem is still there and what it prints out after processing the config file?

rosecitytransit, 2023-06-10:
For now, I added some logging to `System_impl::get_next_control_channel()` since I'm thinking weirdness should show in there:
```
double System_impl::get_next_control_channel() {
  BOOST_LOG_TRIVIAL(info) << "[" << this->get_short_name() << "/" << short_name << "] old currrent_control_channel=" << current_control_channel << ", control_channels.size()=" << control_channels.size() << ", control_channels[]=" << this->control_channels[current_control_channel];
  current_control_channel++;
  if (current_control_channel >= control_channels.size()) {
    current_control_channel = 0;
  }
  BOOST_LOG_TRIVIAL(info) << "[" << this->get_short_name() << "/" << short_name << "] currrent_control_channel=" << current_control_channel << ", control_channels[]=" << this->control_channels[current_control_channel];
  return this->control_channels[current_control_channel];
}
```
When I have time (I'm leaving for the weekend) and get every thing set up on the pi, or when I next need to restart the program, I can definitely try out your change. As well as see if I can induce the problem and further troubleshoot.

taclane, 2023-06-10:
In the event there's an issue mixing up the channels sometime after initial t-r setup, a patch-free option is also to check out the status server plugin.  

The html tables are a little flaky, but it it should still output each system's control channel list, up-to-date as of that moment the page is refreshed.

rosecitytransit, 2023-06-11:
Well it did it again, and the added logging shows control_channels.size()
magically growing from 8 to 10 and then 12, and hours and many retunes
after start.

For now, I've just removed all but the current one in the config. They
don't change that often anyways.

robotastic, 2023-06-21:
I just tried to reproduce this but didn't have any luck. I have one system locked onto a control channel and the other is retuning. Could you share you config file?

```
{
    "ver": 2,
    "sources": [
        {
            "center": 855700000,
            "rate": 2048000,
            "error": 0,
            "gain": 24,
            "digitalRecorders": 1,
            "driver": "osmosdr",
            "device": "rtl=200,buflen=65536"
        }
    ],
    "systems": [

        {
            "control_channels": [
                855462500,
                855237500
            ],
            "type": "p25",
            "modulation": "qpsk",
            "shortName": "test2"
        },
        {
            "control_channels": [
                855560001,
                855570001
            ],
            "type": "p25",
            "modulation": "qpsk",
            "shortName": "test1"
        }
    ],
    "controlWarnRate": -1
}
```

rosecitytransit, 2023-06-22:
I put a copy here: http://www.rosecitytransit.org/files/config.json

Note that I don't think I've had the issue since I went and removed all the other frequencies for the system it was happening on. I may try to do some troubleshooting myself, or close this issue for now and reopen it if I see it happen again.

tadscottsmith, 2024-06-05:
You have a control channel for Pete's Mountain (770.093750) listed in your config for Mount Scott. If trunk-recorder tunes to that control channel it's going to automatically add the alternate CCs that are being broadcast for Pete's Mountain.

rosecitytransit, 2024-06-05:
Don't think I'd get Pete's Mountain from where I'm located, and don't think that there should be any shared frequencies with the Oregon State Radio Project system, but that's a good catch. I know the DB doesn't have all the frequencies listed for Mt Scott.

As I said in the other issue, I could test a more current version and see if the issue still exists.

robotastic, 2024-12-14:
Closing for now - please reopen if it is still a problem

closed 2024-12-14 by robotastic