TrunkRecorder/trunk-recorder #827: TR no longer compiles on Debian 10 with commit 6d68897cc3b8cfbadbb65eeea8f76704a6382015 to current

kb2ear, 2023-06-14:
Trunk-recorder will no longer compile after commit 6d68897cc3b8cfbadbb65eeea8f76704a6382015 on Debian 10

```
[  1%] Checking the git repository for changes...
[  1%] Built target check_git
Scanning dependencies of target trunk_recorder_library
[  1%] Building CXX object CMakeFiles/trunk_recorder_library.dir/trunk-recorder/recorders/recorder.cc.o
In file included from /root/trunk-recorder/trunk-recorder/recorders/recorder.h:59,
                 from /root/trunk-recorder/trunk-recorder/recorders/recorder.cc:1:
/root/trunk-recorder/trunk-recorder/recorders/../gr_blocks/transmission_sink.h:55:9: error: ‘chrono’ in namespace ‘std’ does not name a type
    std::chrono::time_point<std::chrono::steady_clock> d_last_write_time;
         ^~~~~~
/root/trunk-recorder/trunk-recorder/recorders/../gr_blocks/transmission_sink.h:55:4: note: suggested alternative: ‘conj’
    std::chrono::time_point<std::chrono::steady_clock> d_last_write_time;
    ^~~
    conj
/root/trunk-recorder/trunk-recorder/recorders/../gr_blocks/transmission_sink.h:148:9: error: ‘chrono’ in namespace ‘std’ does not name a type
    std::chrono::time_point<std::chrono::steady_clock> get_last_write_time();
         ^~~~~~
/root/trunk-recorder/trunk-recorder/recorders/../gr_blocks/transmission_sink.h:148:4: note: suggested alternative: ‘conj’
    std::chrono::time_point<std::chrono::steady_clock> get_last_write_time();
    ^~~
    conj
make[2]: *** [CMakeFiles/trunk_recorder_library.dir/build.make:63: CMakeFiles/trunk_recorder_library.dir/trunk-recorder/recorders/recorder.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:110: CMakeFiles/trunk_recorder_library.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

kb2ear, 2023-06-14:
Also broken on Ubuntu 18.04.

robotastic, 2024-12-14:
I think this may have been fixed, please reopen if it hasn't

closed 2024-12-14 by robotastic