TrunkRecorder/trunk-recorder #830: Problem to send audio from Trunk-recorder" to "Rdio-Scanner"

py2pe, 2023-06-23:
The First I want to thank you very much the developers. 
I'm having fun in my spare time playing with "Trunk-recorder" and "Rdio-Scanner".
I have a problem, that the audios are only sent to the Rdio-Scanner" when I stop the application "Trunk-Recorder". The audio is not sent in real time. And when I stop (ctrl+c) it goes all at once, and I receiver this mensagem: "Rdio Scanner Upload Success - file size: 24375".
Has anyone figured out why this happens?
Sorry for my English, I'm using google to translate.
this is my config:

{
  "ver": 2,
  "sources":   [
        {
        "center": 171000000,
        "rate": 4000000,
        "error": 0,
        "ppm": -10,
        "digitalRecorders": 8,
        "analogRecorders": 8,
        "driver": "osmosdr",
        "device": "soapy,driver=sdrplay,agc_setpoint=-10,serial=B0002P0007"
        }
               ],
    "systems": [
        {
        "channels": [171306100, 171506250, 172218400],
        "type": "conventionalP25",
        "shortName": "SP-P25-COMERC-VHF",
        "modulation": "fsk4",
        "squelch": -45
        }
    ],
    "captureDir": "/root/trunk-recorder/",
    "plugins": [
              {
          "name": "rdioscanner_uploader",
          "library": "librdioscanner_uploader.so",
          "server": "http://187.50.8.182:3000",
          "systems": [{
                  "shortName": "SP-P25-COMERC-VHF",
                  "apiKey": "SP-P25-COMERC-VHF123",
                  "systemId": 65
                     }]
              }
             ]
}

Regards,

Dewey3, 2023-06-25:
How are you sending the audio to Rdio-Scanner?  When I run Trunk-Recorder and Rdio-Scanner on the same system (RPi), I use Rdio-Scanner's directory watch, and when I run Trunk-Recorder and Rdio-Scanner on separate systems, I let Trunk-Recorder send the audio to Rdio-Scanner with the Trunk-Recorder Rdio-Scanner plugin.  I have never had a problem either of these two ways.

I run three different Trunk-Recorders on three different RPis and used to run a local instance of Rdio-Scanner's Directory Watch on each RPi and let each RPi Rdio-Scanner session send the ingested audio to a "central" Rdio-Scanner, but that required MUCH more maintenance and was with earlier Trunk-Recorder versions before I learned the plug-in, but that way also worked without any problems.

Dewey3, 2023-06-25:
I'm not sure if this is the problem, but it looks like we run the same way, but your config file is a little different than mine.  Try moving "captureDir" up under "ver" and before "sources".  Again, I don't know if this will solve the problem, but it is the biggest difference between your config and mine.

py2pe, 2023-06-25:
Hi Dewey3

Thank you very very much for reply.

I'm not sure what happened, but right now it's working. Out of nowhere it stops working and it no longer sends to the radio-scanner server. I'll try your suggestion.
I am attaching two files of my setup.

[http://187.50.8.182:3000/](url)
[image]
[image]

py2pe, 2023-06-26:
The problem came back, and I did your recommendation.
In the attached video, notice that the audio is only sent to rdio-scanner when I stop running trunk-recorder.
Thanks for helping me, I still need to figure out what it could be. This is a case that was working before, and it stopped out of nowhere. The other computers are sending normal audio.

https://github.com/robotastic/trunk-recorder/assets/134792784/191a5b7d-74e7-42a2-acf0-42a2beb90868

Regards

frameshift18, 2023-06-26:
The audio that is all sent at the end, is it a bunch of separate calls all in one long file? I am not sure how ConventionalDMR works but if its similar to normal analog conventional you need to make sure your squelch is set correctly.

py2pe, 2023-06-26:
Hi frameshift18,

Thanks for you reply.

Now I am working. Later I will be at my house and I will take the test as per your recommendation. Thank you one more time.

closed 2024-12-14 by robotastic