TrunkRecorder/trunk-recorder #836: Newest Version Continues to Fumble UIDs

Dewey3, 2023-07-16:
This is a continuation from https://github.com/robotastic/trunk-recorder/issues/760, which I hope is ok to re-open here since the original is now so far down the list.  (If it is a problem, my sincerest apologies and I will delete this issue)  Anyway, I just finished building

```
Trunk-Recorder: 4.6.0
commit 767455ba297292ec563676d319c80f9f213f5f67 (master)
describe LTS-219-g767455b
Author: tadscottsmith <78445808+tadscottsmith@users.noreply.github.com>
Date: 2023-07-12 11:49:39 -0400
Include TDMA slot number in transmission_sink(). (#835)
```

and believe it or not, it is still missing the transmitting UIDs when transmissions are right behind each other.  What happens is that while the conversation itself continues fine, the UID does not change, but continues to show the first/prior UID.  The last time the UID changes worked correctly on the very busy Prince George's Co, MD system (https://www.radioreference.com/db/sid/6341) was version

```
Trunk-Recorder: 4.3.2
commit 10a276385124ef48e7ad6a3493236082c360ec0e (master)
describe 2.1.2-1474-g10a2763
Author: Luke Berndt <lukekb@gmail.com>
Date: 2022-09-15 22:19:36 -0400
Update README.md
```

when @tadscottsmith figured out the problem, and contributed a fix.  While I will try the new updates every few releases, I still run the 9/15/2022 v4.3.2 version since it correctly keeps up with the UIDs no matter how busy the system.  Please don't take this as a complaint as I can keep running the 9/15/2022 v4.3.2 as I do now, but I just want those on the coding side to know that I still see this issue.  I'm sorry that I can't do more in helping solve it, but the coding and understanding the code parts are well above my feeble brain.

robotastic, 2023-07-17:
Thanks for capturing this - the good news is that I can sort of receive the PGC system in DC, so I can try doing some debugging.

theficus, 2023-08-09:
I can help verify any fixes here as well. The system I scan here in Seattle is also P25P2 and is super busy (I have to run 16 recorders just to keep up). I've also observed similar behavior.

tadscottsmith, 2023-08-10:
Can someone grab like 30 seconds of I/Q recording where this is happening? I can look some more, but don't have access to a P25P2 system.

Dewey3, 2023-08-10:
> Can someone grab like 30 seconds of I/Q recording where this is happening? I can look some more, but don't have access to a P25P2 system.

I don't know how to create/generate an IQ file, but willing to try to if given the steps.  I can also install the latest TrunkRecorder and open up the RPi for testing the way you did last time if that is better, or at least helps.

xmleger, 2023-09-03:
Not sure if this is related to a similar behavior that I have been seeing with my analog trunking (Smartnet) and analog conventional channel systems. They seem to capture the source (UID) info for the first transmitting unit, but none of the following source data is listed in the json for additional unit transmissions on that same audio capture file. The P25 conventional system I have running DOES capture all UIDs for transmissions. 

Looks like I am currently running commit: 2c7a39f0fc67d299b100221e276d04798823a2cf

I tried running on a newer commit with similar behavior noted.

tadscottsmith, 2024-01-09:
@Dewey3 There's been a lot of work done on a preview branch for v5.0. One of the things being worked on is a fix that should hopefully address this issue. If you have some time, would you be willing to test the `test/tdu-test` branch to see if it helps address this issue? I know you monitor a very buy system that frequently sees this issue.

I believe something similar to this would help you build a system on the test branch. 

```
mkdir trunk-build
git clone https://github.com/robotastic/trunk-recorder.git
cd trunk-recorder
git checkout test/tdu-test
cd ../trunk-build
cmake ../trunk-recorder
make -j$(nproc)
```

theficus, 2024-01-09:
I'll give this a try as well as I monitor a very busy system too (tens of thousands of calls a day) and have observed this issue.

theficus, 2024-01-09:
> I'll give this a try as well as I monitor a very busy system too (tens of thousands of calls a day) and have observed this issue.

Well... that didn't work 🤣.

When I tried this build it just would not lock onto a control signal. I had to revert back to the master branch to get recordings working again. I've attached a log in case there's anything useful here.

[01-08-2024_1937_00 copy.txt](https://github.com/robotastic/trunk-recorder/files/13868591/01-08-2024_1937_00.copy.txt)

taclane, 2024-01-09:
> Well... that didn't work 🤣.

```
[2024-01-08 19:37:30.494182] (info)   Rate: 2148000
...
[2024-01-08 19:37:31.625494] (info)   	 Xlating Channelizer single-stage decimator - Decim: 89 Resampled Rate: 24134.8 Lowpass Size: 431
[2024-01-08 19:37:31.625746] (info)   	 Channelizer ARB - Symbol Rate: 24000 Resampled Rate: 24134.8 ARB Rate: 0.994413
```
Would you be willing to try running this at a sample rate of `2160000` or `2400000`?  

It's possible rates that can't cleanly divide down in the channelizer end up having problems.  As a quick test, I set one of mine to `2485000` and it suddenly had issues receiving on the control channel.

It shouldn't at all be a concern to bump up the signal rate.  Part of the magic in the 5.0 improvements is that it should be a lot more efficient with a 2.4 MHz source than 4.7 could be with a 2.148 MHz source.

theficus, 2024-01-09:
Sure, I'll give it a shot. I just can't go above 2400000 because any higher and my RTL-SDRs crap out. :)

taclane, 2024-01-09:
Either 2160000 or 2400000 is fine.  

The goal is mostly to just get an even multiple of 24000, but both of those are "standard" rates generally suggested by RTL drivers.

Dewey3, 2024-01-09:
Thanks @tadscottsmith and @taclane !  I may not be able to do a complete test until this weekend, but I hope to install this build when I get home today.  Sorry for the delay until the weekend (day job), but I will be back to you both.

Dewey3, 2024-01-11:
I'm having the same problem... I can't get a lock on the control channel.  I've tried 240000 and 216000, no joy.  I even tried 2040000 since that is divisible by 24000 and I believe close to the Nooelec RTL-SDR dongle 2.4 MSPS sample rate (if that matters).

theficus, 2024-01-11:
For what it's worth I'm also using Nooelec RTL-SDR dongles. Very strange issue.

tadscottsmith, 2025-10-30:
@robotastic can we setup a dev branch for this? I think I found a pretty simple fix, but I'd like for users to be able to test on a live P25 P2 systems instead of just the 30 seconds of IQ I have.

robotastic, 2025-10-30:
Awesome - Just created a branch here: https://github.com/TrunkRecorder/trunk-recorder/tree/dev/id-fix

tadscottsmith, 2025-10-30:
@Dewey3 and @theficus - I've merged the proposed fixes into the `dev/id-fix` branch. 

If you're comfortable building from source, I'd greatly appreciate help testing these changes on an actual P25 Phase 2 system.

theficus, 2025-10-30:
Sure. I'll give it a shot this weekend. I capture a super busy system 800+ calls an hour during peak traffic), so this should be a good stress test.

tadscottsmith, 2025-10-30:
> It's nowhere near as bad as before, but I am still seeing the problem on https://www.radioreference.com/db/sid/6341.

Are you comparing to RDIO or what's in the JSON files? I don't know that RDIO is always updating as fast as what's actually sent in the JSON.

theficus, 2025-10-31:
I've let it run for a little while. It doesn't seem any worse than before but I'm still seeing some instances of log entries like this:

> 2025-10-30 18:30:50.606756] (info)   [psern025]        421C     TG:       3931 (               M.A.R.S.)        Freq: 853.525000 MHz     GROUP MISMATCH -  Recorder TG: 3931 Received TG: 207 Recorder state: idle incoming: 1280

cschmittiey, 2025-10-31:
hey i also monitor psern and see a lot of the same thing. happy to share logs/debug, probably easier to ping me in the discord.

tadscottsmith, 2025-11-03:
> I've let it run for a little while. It doesn't seem any worse than before but I'm still seeing some instances of log entries like this:

Group mismatch seems more indicative of a signal issue, but could be a different bug as well. What's your control channel msgs/sec rate, and how often do you see these messages?

Dewey3, 2025-11-14:
My apologies, life happened so I had to redirect my attention for a couple weeks.  I'm still looking at the Rdio output and have noticed that while the unchanging UIDs still happen, but **no where as much** in this test version, they still happen.  However, I did notice that it seems to always be when the field units are talking.  Anytime a dispatcher transmits, the UID immediately changes to the dispatcher UID, even if the dispatcher appears to cut in while the field unit is still talking.

I do have a quick question.  I'm not well versed on how building/compiling a github project works, but I am now using a version that is built using the latest 11/12/2025, v5.1.1 build, then adding your `dev/id-fix` dev branch.  When compiling, does the dev branch checkout cause the build to use the code that was current when the dev branch was created, or will it use the latest available master branch code?

SimonSheehan, 2026-02-01:
The changes overall have made a good difference for me - not always perfect but, a lot more hits than missing.

I'm wondering if we can merge the current `dev/id-fix` in for now but keep this issue open. It's certainly working code that helps.