TrunkRecorder/trunk-recorder #838: Is this a known bug? trunk-recorder just stops saving files?

bnelson333, 2023-07-29:
I don't honestly know which version of trunk-recorder I have installed, nor do I know how to check. But I've recently noticed that the implementation on my linux box just randomly stops recording. trunk-recorder is still running, but it stops dumping files to the output directory. Is that a known issue? Will upgrading help? Or is this something new that nobody's noticed before? To handle it right now I have a script parsing the output directory checking for a stop in new file creation (by checking last modified date of files), if it senses no new files after a few minutes it'll reboot. But obviously that's not ideal, has that been noticed or fixed recently?

robotastic, 2023-07-29:
Are you trying to record a SmartNet system? This is sort of a known issue. There is a setup to automatically restart Trunk Recorder if the it stops receiving a system. I have noticed that this could be a sign the things may not be tuned that well. (Check your Error setting and Gain)

rosecitytransit, 2023-07-29:
Have you looked at the output logs? Does it still appear to be decoding and processing?

bnelson333, 2023-07-29:
what's strange is I have two identical machines doing this, they're both an optiplex 790 with 2 nesdr smart v4 dongles (my system requires 2). all the config and everything is exactly the same between the two (I literally copy/pasted). One of the machines doesn't ever do it, and one machine does it about every day. The one that does it every day has a later version of trunk-recorder installed. But again, I don't know how to check which it is. I built the second system maybe a year later? So whatever it pulled down from git at the time is what it has. I'm temped to re-build it with a new version but didn't want to bother if the new version is going to do the same thing. Is it possible to build an old version? I tried to copy the trunk-build and trunk-recorder dirs from machine #1 to #2 but it wouldn't run, so I'm not sure what else it does in the build process.

taclane, 2023-07-29:
As long as it isn't *very* old, you can check what version of trunk recorder you have with:
`trunk-recorder -v`

I experienced something similar with trunk-recorder 4.6.0, but if you can use git to [update and install](https://github.com/robotastic/trunk-recorder/blob/master/docs/INSTALL-LINUX.md), things should be good again after some commits around June 21. 

What I saw was trunk-recorder attempting to retune the control channel and then soft-locking indefinitely.  No further console messages and CPU usage dropped down to a few percent; trunk-recorder was "running" but stopped doing anything.  If that sounds like you, try using the above instructions to get a fresh update of trunk-recorder from github.

robotastic, 2024-12-14:
Closing - reopen if this is still an issue

closed 2024-12-14 by robotastic