TrunkRecorder/trunk-recorder #848: audioplayer.php fails on php 8.2

jeffbarclay, 2023-10-24:
the strftime is depicted in newer php versions

[Tue Oct 24 08:25:49 2023] PHP Warning:  Cannot modify header information - headers already sent by (output started at /Users/me/gr-osmosdr/audioplayer.php:126) in /Users/me/gr-osmosdr/audioplayer.php on line 155
[Tue Oct 24 08:25:49 2023] [::1]:56733 [200]: GET /audioplayer.php?since=0&tg=&date=

jeffbarclay, 2023-10-24:
not the most eloquent time format but at least this worked:

ADD:
date_default_timezone_set('America/New_York');
$date = new DateTimeImmutable();

REPLACED:
           'date'  => strftime('%F %T', $TIME),
WITH:
           'date'   => date("Y-m-d\TH:i:s\Z",$TIME),

** would like the Z to show milliseconds

Not sure what the following does, but times are correct on filtered list???

 if ($filter_date->format('Y-m-d') != strftime('%F', $TIME))
                {
                    continue;
                }

robotastic, 2024-12-14:
Thanks for the fix - this player is not really supported, but I added these changes in

closed 2024-12-14 by robotastic