TrunkRecorder/trunk-recorder #849: Airspy R2 Serial Number

jeffbarclay, 2023-10-26:
it seems I need to have the serial number to assign to device?
Im using a Mac (Intel) and have no way to get the serial number via airspy 

device="airspy" 
or
device="airspy=0"

both do not work??? I can connect with Gqrx and radio works fine... any ideas if S/N is needed or another way to work with trunk-recorder

wx-jc, 2023-10-27:
Do you have your json formatting correct?  Here is my airspy config that works fine (on Linux and Mac M1):

"ver": 2,
        "sources": [{
                "center": 854543750,
                "rate": 10000000,
                "ifGain": 7,
                "mixGain": 7,
                "lnaGain": 7,
                "digitalRecorders": 15,
                "driver": "osmosdr",
                "device": "airspy=0",
                "ppm": 1.1

jeffbarclay, 2023-10-27:
still no luck:
```
{
    "ver": 2,
    "captureDir":"/Users/jeffreybarclay/gr-osmosdr",
    "sources": [{
        "center": 853687500,
        "rate": 1048000,
        "ifGain": 7,
		"mixGain": 7,
		"lnaGain": 7,
		"digitalRecorders": 15,
        "driver": "osmosdr",
        "device": "airspy"
    }],
    "systems": [{
        "control_channels": [853687500, 853862500, 853887500, 853937500],
        "type": "p25",
        "talkgroupsFile": "/users/jeffreybarclay/gr-osmosdr/mocotg.csv",
        "shortName": "montgomery",
        "modulation": "qpsk"
    }]
    
}
```
```

[2023-10-27 08:58:54.918276] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------
[2023-10-27 08:58:54.918323] (info)   

-------------------------------------
INSTANCE
-------------------------------------
[2023-10-27 08:58:54.918346] (info)   Using Config file: MoCo-spy.json
[2023-10-27 08:58:54.918354] (info)   Trunk-Recorder: Version: 4.6.0
[2023-10-27 08:58:54.918361] (info)   Log to File: false
[2023-10-27 08:58:54.918369] (info)   Log Directory: logs
[2023-10-27 08:58:54.918450] (info)   Temporary Transmission Directory: /Users/jeffreybarclay/gr-osmosdr
[2023-10-27 08:58:54.918497] (info)   Capture Directory: /Users/jeffreybarclay/gr-osmosdr
[2023-10-27 08:58:54.918508] (info)   Upload Server: 
[2023-10-27 08:58:54.918516] (info)   Broadcastify Calls Server: 
[2023-10-27 08:58:54.918524] (info)   Status Server: 
[2023-10-27 08:58:54.918531] (info)   Instance Key: 
[2023-10-27 08:58:54.918538] (info)   Instance Id: 
[2023-10-27 08:58:54.918545] (info)   Broadcast Signals: false
[2023-10-27 08:58:54.918553] (info)   Default Mode: digital
[2023-10-27 08:58:54.918563] (info)   Call Timeout (seconds): 3
[2023-10-27 08:58:54.918573] (info)   Control channel warning rate: 10
[2023-10-27 08:58:54.918584] (info)   Control channel retune limit: 0
[2023-10-27 08:58:54.918595] (info)   Phase 1 Software Vocoder: false
[2023-10-27 08:58:54.918606] (info)   Enable Audio Streaming: false
[2023-10-27 08:58:54.918619] (info)   Record Unit to Unit Voice Calls: true
[2023-10-27 08:58:54.918631] (info)   New Call from UPDATE Messagestrue
[2023-10-27 08:58:54.918645] (info)   Frequency format: mhz
[2023-10-27 08:58:54.918657] (info)   Status as String: true
[2023-10-27 08:58:54.918674] (info)   Log Level: info
[2023-10-27 08:58:54.918689] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
[2023-10-27 08:58:54.918722] (info)   

System Number: 1
-------------------------------------
[2023-10-27 08:58:54.918734] (info)   Short Name: montgomery
[2023-10-27 08:58:54.918775] (info)   System Type: p25
[2023-10-27 08:58:54.918801] (info)   Control Channels: 
[... 45 lines cut ...]
[2023-10-27 08:58:54.920312] (info)   Rate: 1048000
[2023-10-27 08:58:54.920325] (info)   Error: 0
[2023-10-27 08:58:54.920336] (info)   PPM Error: 0
[2023-10-27 08:58:54.920348] (info)   Auto gain control: false
[2023-10-27 08:58:54.920361] (info)   Gain: 0
[2023-10-27 08:58:54.920373] (info)   IF Gain: 7
[2023-10-27 08:58:54.920383] (info)   BB Gain: 0
[2023-10-27 08:58:54.920391] (info)   LNA Gain: 7
[2023-10-27 08:58:54.920400] (info)   PGA Gain: 0
[2023-10-27 08:58:54.920409] (info)   TIA Gain: 0
[2023-10-27 08:58:54.920421] (info)   MIX Gain: 7
[2023-10-27 08:58:54.920429] (info)   AMP Gain: 0
[2023-10-27 08:58:54.920437] (info)   VGA Gain: 0
[2023-10-27 08:58:54.920444] (info)   VGA1 Gain: 0
[2023-10-27 08:58:54.920452] (info)   VGA2 Gain: 0
[2023-10-27 08:58:54.920459] (info)   Idle Silence: false
[2023-10-27 08:58:54.920467] (info)   Digital Recorders: 15
[2023-10-27 08:58:54.920476] (info)   SigMF Recorders: 0
[2023-10-27 08:58:54.920484] (info)   Analog Recorders: 0
[2023-10-27 08:58:54.920499] (info)   Source Device: airspy=0
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.7.0
built-in source types: file rtl rtl_tcp uhd rfspace soapy redpitaya 
[INFO] [UHD] Mac OS; Clang version 14.0.3 (clang-1403.0.22.14.1); Boost_108200; UHD_4.5.0.HEAD-0-g471af98f
[2023-10-27 08:58:55.549844] (error)   Failed parsing Config: No supported devices found (check the connection and/or udev rules).
```

gofaster, 2023-10-27:
I no longer have any Airspys in the mix but when I did a few years ago, this is what worked.  This was running the patched osmocom driver with multiple Airspy support.  Add the serial number as a parameter.
```
      "driver": "osmosdr",
      "device": "airspy=0x644064DC236BE64EA"
```

wx-jc, 2023-10-27:
That rate is not valid for Airspy.  If it's an Airspy R2, your two choices are 2500000 or 10000000 (2.5 MHz or 10 MHz).  I don't know if that will solve all the problems, as you may also need a udev rule added...but try that first.

jeffbarclay, 2023-10-27:
yea just tried both of those with same error...

also tried changing to serial # : airspy=0x26A464DC284F2C93

same error, don't know anything about udev? researching now!

jeffbarclay, 2023-10-27:
this same config file executes fine with previous settings and device=rtl=00000001 and a cheaper sdr plugged in...

taclane, 2023-10-27:
Osmosdr may not be fine.  It doesn't list airspy as an available driver.
`built-in source types: file rtl rtl_tcp uhd rfspace soapy redpitaya`

If gr-osmosdr cannot detect the airspy libraries during the `cmake` step when installing it, trunk-recorder will not be able to use them.  You should see something like this after the cmake is run, showing you what devices it is enabled for:
```
-- ######################################################
-- # Gnuradio enabled components
-- ######################################################
--   * Python support
--   * IQ File Source & Sink
--   * Osmocom RTLSDR
--   * RTLSDR TCP Client
--   * Ettus USRP Devices
--   * HackRF & rad1o Badge
--   * RFSPACE Receivers
--   * AIRSPY Receiver
--   * SoapySDR support
--   * Red Pitaya SDR
```
Depending on what method you're using to install everything, it may be as simple as a `brew install airspy`, recompiling gr-osmosdr, then recompiling trunk-recorder.  You'll know for sure that things are lined up correctly when that `built-in source types` line on startup also includes `airspy`.

jeffbarclay, 2023-10-27:
I did just install airspy via homebrew yesterday to get the airspy_info to tell me the serial number... So just re-install everything through homebrew may be the fix... Im new to homebrew (this was my first project) do I need to delete everything and start over... uggg! or can I just re-install?

taclane, 2023-10-27:
You're fine.  All you're essentially going to do is rebuild gr-osmosdr (with airspy drivers), install it on top of whatever you did previously, then do the same with trunk-recorder.

If you're following along the trunk-recorder mac homebrew directions, all you should really need to do is restart from the middle of the "Install the OsmoSDR Package for GNURadio" step.  `cd` into the `build` directory, do a `make clean` and the continue on from the `cmake ..` step. After the cmake, you should see the airspy drivers in the "Gnuradio enabled components" printout.

Afterwards, do the same general thing with trunk-recorder.  Clean out the build directory, then resume from the cmake step.

jeffbarclay, 2023-10-27:
can't thank you enough - its been days trying to troubleshoot this... the re-install worked only really needed to re-intall the osmosdr section but its receiving control channel... needs some tuning to pickup usable TGs

taclane, 2023-10-27:
Great!  Glad it works!  👍 

You should be able to close out this issue for now.

jeffbarclay, 2023-11-28:
trying to duplicate install on Raspberry pi and getting the same Airspy error on no library - not using homebrew so I'm looking for how to install drivers/libraries on the pi

-------------------------------------
     Trunk Recorder
-------------------------------------
[2023-11-28 16:23:04.216503] (info)   

-------------------------------------
INSTANCE
-------------------------------------
[2023-11-28 16:23:04.216560] (info)   Using Config file: ./config.json
[2023-11-28 16:23:04.216638] (info)   Trunk-Recorder: Version: 4.7.0
[2023-11-28 16:23:04.216670] (info)   Log to File: false
[2023-11-28 16:23:04.216717] (info)   Log Directory: logs
[2023-11-28 16:23:04.216823] (info)   Temporary Transmission Directory: /dev/shm
[2023-11-28 16:23:04.216898] (info)   Capture Directory: ~/trunk-build
[2023-11-28 16:23:04.216963] (info)   Upload Server: 
[2023-11-28 16:23:04.217026] (info)   Broadcastify Calls Server: 
[2023-11-28 16:23:04.217089] (info)   Status Server: 
[2023-11-28 16:23:04.217165] (info)   Instance Key: 
[2023-11-28 16:23:04.217228] (info)   Instance Id: 
[2023-11-28 16:23:04.217291] (info)   Broadcast Signals: false
[2023-11-28 16:23:04.217354] (info)   Default Mode: digital
[2023-11-28 16:23:04.217417] (info)   Call Timeout (seconds): 3
[2023-11-28 16:23:04.217491] (info)   Control channel warning rate: 10
[2023-11-28 16:23:04.217554] (info)   Control channel retune limit: 0
[2023-11-28 16:23:04.217617] (info)   Phase 1 Software Vocoder: false
[2023-11-28 16:23:04.217679] (info)   Enable Audio Streaming: false
[2023-11-28 16:23:04.217740] (info)   Record Unit to Unit Voice Calls: true
[2023-11-28 16:23:04.217803] (info)   New Call from UPDATE Messagestrue
[2023-11-28 16:23:04.217874] (info)   Frequency format: mhz
[2023-11-28 16:23:04.217936] (info)   Status as String: true
[2023-11-28 16:23:04.217998] (info)   Log Level: info
[2023-11-28 16:23:04.218066] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
[2023-11-28 16:23:04.218239] (info)   

System Number: 1
-------------------------------------
[2023-11-28 16:23:04.218355] (info)   Short Name: montgomery
[2023-11-28 16:23:04.218418] (info)   System Type: p25
[2023-11-28 16:23:04.218480] (info)   Control Channels: 
[2023-11-28 16:23:04.218546] (info)     853.687500 MHz
[2023-11-28 16:23:04.218673] (info)     853.862500 MHz
[2023-11-28 16:23:04.218750] (info)     853.887500 MHz
[2023-11-28 16:23:04.218823] (info)     853.937500 MHz
[2023-11-28 16:23:04.218899] (info)   Loading Talkgroups...
[2023-11-28 16:23:04.218960] (info)   Reading Talkgroup CSV File: xmocotg.csv
[2023-11-28 16:23:04.219871] (info)   Found Columns: Decimal, Hex, Alpha Tag, Mode, Description, Tag
[2023-11-28 16:23:04.220811] (info)   Read 120 talkgroups.
[2023-11-28 16:23:04.220955] (info)   Talkgroups File: xmocotg.csv
[2023-11-28 16:23:04.221037] (info)   Modulation: qpsk
[2023-11-28 16:23:04.221104] (info)   Conversation Mode: true
[2023-11-28 16:23:04.221167] (info)   Analog Recorder Maximum Deviation: 4000
[2023-11-28 16:23:04.221233] (info)   Filter Width: 1
[2023-11-28 16:23:04.221310] (info)   Squelch: -160
[2023-11-28 16:23:04.221373] (info)   API Key: 
[2023-11-28 16:23:04.221429] (info)   Broadcastify API Key: 
[2023-11-28 16:23:04.221464] (info)   Broadcastify Calls System ID: 0
[2023-11-28 16:23:04.221511] (info)   Upload Script: 
[2023-11-28 16:23:04.221572] (info)   Compress .wav Files: true
[2023-11-28 16:23:04.221636] (info)   Call Log: true
[2023-11-28 16:23:04.221698] (info)   Audio Archive: true
[2023-11-28 16:23:04.221760] (info)   Transmission Archive: true
[2023-11-28 16:23:04.221821] (info)   Loading Unit Tags...
[2023-11-28 16:23:04.221881] (info)   Unit Tags File: 
[2023-11-28 16:23:04.221943] (info)   Record Unknown Talkgroups: false
[2023-11-28 16:23:04.222005] (info)   Decode MDC: false
[... 18 lines cut ...]
[2023-11-28 16:23:04.223058] (info)   Driver: osmosdr
[2023-11-28 16:23:04.223122] (info)   Center: 853.500000 MHz
[2023-11-28 16:23:04.223218] (info)   Rate: 10000000
[2023-11-28 16:23:04.223293] (info)   Error: 0
[2023-11-28 16:23:04.223358] (info)   PPM Error: 0
[2023-11-28 16:23:04.223422] (info)   Auto gain control: false
[2023-11-28 16:23:04.223483] (info)   Gain: 0
[2023-11-28 16:23:04.223544] (info)   IF Gain: 7
[2023-11-28 16:23:04.223605] (info)   BB Gain: 0
[2023-11-28 16:23:04.223665] (info)   LNA Gain: 7
[2023-11-28 16:23:04.223726] (info)   PGA Gain: 0
[2023-11-28 16:23:04.223786] (info)   TIA Gain: 0
[2023-11-28 16:23:04.223846] (info)   MIX Gain: 7
[2023-11-28 16:23:04.223907] (info)   AMP Gain: 0
[2023-11-28 16:23:04.223967] (info)   VGA Gain: 0
[2023-11-28 16:23:04.224027] (info)   VGA1 Gain: 0
[2023-11-28 16:23:04.224087] (info)   VGA2 Gain: 0
[2023-11-28 16:23:04.224147] (info)   Idle Silence: 0
[2023-11-28 16:23:04.224207] (info)   Digital Recorders: 15
[2023-11-28 16:23:04.224269] (info)   SigMF Recorders: 0
[2023-11-28 16:23:04.224330] (info)   Analog Recorders: 0
[2023-11-28 16:23:04.224398] (info)   Source Device: airspy=0x26A464DC284F2C93
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx 
[2023-11-28 16:23:04.241789] (error)   Failed parsing Config: Failed to open AirSpy device (-5) AIRSPY_ERROR_NOT_FOUND

jeffbarclay, 2023-11-28:
after reboot now getting different error?

-------------------------------------
     Trunk Recorder
-------------------------------------
[2023-11-28 16:36:59.271494] (info)   

-------------------------------------
INSTANCE
-------------------------------------
[2023-11-28 16:36:59.271652] (info)   Using Config file: ./config.json
[2023-11-28 16:36:59.271711] (info)   Trunk-Recorder: Version: 4.7.0
[2023-11-28 16:36:59.271765] (info)   Log to File: false
[2023-11-28 16:36:59.271842] (info)   Log Directory: logs
[2023-11-28 16:36:59.272124] (info)   Temporary Transmission Directory: /dev/shm
[2023-11-28 16:36:59.272257] (info)   Capture Directory: ~/trunk-build
[2023-11-28 16:36:59.272305] (info)   Upload Server: 
[2023-11-28 16:36:59.272346] (info)   Broadcastify Calls Server: 
[2023-11-28 16:36:59.272384] (info)   Status Server: 
[2023-11-28 16:36:59.272422] (info)   Instance Key: 
[2023-11-28 16:36:59.272459] (info)   Instance Id: 
[2023-11-28 16:36:59.272495] (info)   Broadcast Signals: false
[2023-11-28 16:36:59.272533] (info)   Default Mode: digital
[2023-11-28 16:36:59.272586] (info)   Call Timeout (seconds): 3
[2023-11-28 16:36:59.272847] (info)   Control channel warning rate: 10
[2023-11-28 16:36:59.272923] (info)   Control channel retune limit: 0
[2023-11-28 16:36:59.272986] (info)   Phase 1 Software Vocoder: false
[2023-11-28 16:36:59.273049] (info)   Enable Audio Streaming: false
[2023-11-28 16:36:59.273112] (info)   Record Unit to Unit Voice Calls: true
[2023-11-28 16:36:59.273174] (info)   New Call from UPDATE Messagestrue
[2023-11-28 16:36:59.273250] (info)   Frequency format: mhz
[2023-11-28 16:36:59.273312] (info)   Status as String: true
[2023-11-28 16:36:59.273374] (info)   Log Level: info
[2023-11-28 16:36:59.273444] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
[2023-11-28 16:36:59.273644] (info)   

System Number: 1
-------------------------------------
[2023-11-28 16:36:59.273790] (info)   Short Name: montgomery
[2023-11-28 16:36:59.273859] (info)   System Type: p25
[2023-11-28 16:36:59.273921] (info)   Control Channels: 
[2023-11-28 16:36:59.273981] (info)     853.687500 MHz
[2023-11-28 16:36:59.274099] (info)     853.862500 MHz
[2023-11-28 16:36:59.274176] (info)     853.887500 MHz
[2023-11-28 16:36:59.274248] (info)     853.937500 MHz
[2023-11-28 16:36:59.274324] (info)   Loading Talkgroups...
[2023-11-28 16:36:59.274384] (info)   Reading Talkgroup CSV File: xmocotg.csv
[2023-11-28 16:36:59.278148] (info)   Found Columns: Decimal, Hex, Alpha Tag, Mode, Description, Tag
[2023-11-28 16:36:59.279083] (info)   Read 120 talkgroups.
[2023-11-28 16:36:59.279229] (info)   Talkgroups File: xmocotg.csv
[2023-11-28 16:36:59.279335] (info)   Modulation: qpsk
[2023-11-28 16:36:59.279392] (info)   Conversation Mode: true
[2023-11-28 16:36:59.279442] (info)   Analog Recorder Maximum Deviation: 4000
[2023-11-28 16:36:59.279505] (info)   Filter Width: 1
[2023-11-28 16:36:59.279582] (info)   Squelch: -160
[2023-11-28 16:36:59.279646] (info)   API Key: 
[2023-11-28 16:36:59.279711] (info)   Broadcastify API Key: 
[2023-11-28 16:36:59.279773] (info)   Broadcastify Calls System ID: 0
[2023-11-28 16:36:59.279835] (info)   Upload Script: 
[2023-11-28 16:36:59.279896] (info)   Compress .wav Files: true
[2023-11-28 16:36:59.279957] (info)   Call Log: true
[2023-11-28 16:36:59.280019] (info)   Audio Archive: true
[2023-11-28 16:36:59.280081] (info)   Transmission Archive: true
[2023-11-28 16:36:59.280142] (info)   Loading Unit Tags...
[2023-11-28 16:36:59.280246] (info)   Unit Tags File: 
[2023-11-28 16:36:59.280311] (info)   Record Unknown Talkgroups: false
[2023-11-28 16:36:59.280373] (info)   Decode MDC: false
[... 95 lines cut ...]

[2023-11-28 16:37:45.690022] (info)   

-------------------------------------
PLUGINS
-------------------------------------
[2023-11-28 16:37:45.690107] (info)   Setting up plugin -  Name: openmhz_uploader	 Library file: libopenmhz_uploader.so
[2023-11-28 16:37:45.741190] (info)   Setting up plugin -  Name: broadcastify_uploader	 Library file: libbroadcastify_uploader.so
[2023-11-28 16:37:45.749516] (info)   Setting up plugin -  Name: unit_script	 Library file: libunit_script.so
[2023-11-28 16:37:45.756101] (info)   Setting up plugin -  Name: stat_socket	 Library file: libstat_socket.so
[2023-11-28 16:37:45.791473] (info)   Setting up plugin -  Name: rdioscanner_uploader	 Library file: librdioscanner_uploader.so
[2023-11-28 16:37:45.800405] (info)   Rdio Scanner Server: http://localhost:3000
[2023-11-28 16:37:45.806995] (info)   Uploading calls for: montgomery
[2023-11-28 16:37:45.807149] (info)   Loaded 5 Plugins
[2023-11-28 16:37:45.807238] (info)   

[2023-11-28 16:37:45.809977] (info)   [montgomery]	Started with Control Channel: 853.687500 MHz
[2023-11-28 16:37:45.812690] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 10000000
[2023-11-28 16:37:45.842817] (info)   	 P25 Trunking ARB - Initial Rate: 10000000 Resampled Rate: 25000 Initial Decimation: 100 System Rate: 24000 ARB Rate: 0.96
terminate reached from thread id: f62e5880Got std::runtime_error
boost::thread_resource_error: Resource temporarily unavailable
0xf68d2bc4: (std::vector<gr::tag_t, std::allocator<gr::tag_t> >* std::__do_uninit_fill_n<std::vector<gr::tag_t, std::allocator<gr::tag_t> >*, unsigned int, std::vector<gr::tag_t, std::allocator<gr::tag_t> > >(std::vector<gr::tag_t, std::allocator<gr::tag_t> >*, unsigned int, std::vector<gr::tag_t, std::allocator<gr::tag_t> > const&)+0xdfc)
0xf65f0fec: (std::rethrow_exception(std::__exception_ptr::exception_ptr)+0xa4)
0xf65f1078: (std::terminate()+0xc)
Aborted

closed 2024-12-14 by robotastic