TrunkRecorder/trunk-recorder #850: Curl missing from docker image

arcreigh, 2023-11-03:
Looks like curl is missing from the latest image, I use it for rdio-scanner uploads can we get this re-implemented?

Dewey3, 2023-11-04:
I don't know if it's related, but if it helps whoever ultimately looks at this, I got ton of curl warning when doing a non-Docker build today.  The `make` file log is attached and the curl warnings start at 91%.

[TrunkRecorderInstallationWarnings.txt](https://github.com/robotastic/trunk-recorder/files/13257631/TrunkRecorderInstallationWarnings.txt)

taclane, 2023-11-04:
> I don't know if it's related, but if it helps whoever ultimately looks at this, I got a ton of curl depreciated warnings when doing a non-Docker build today. The `make` file log is attached and the curl warnings start at 91%.
> 
> [TrunkRecorderInstallationWarnings.txt](https://github.com/robotastic/trunk-recorder/files/13257631/TrunkRecorderInstallationWarnings.txt)

That's unrelated to the docker issue.  The curl libraries got an update a little while back, and is recommending that the plugins switch over to some new functions because the methods currently used will stop working someday.  It'll get fixed eventually, but the huge wall of compile warnings can be ignored for now.

> Looks like curl is missing from the latest image, I use it for rdio-scanner uploads can we get this re-implemented?

I can't speak for what's up with curl, but is using the rdio-scanner plugin an option?  As long as you don't have any sort of complicated upload logic, and are using rdio-scanner 6.0 or greater, it seems to be the preferred method over upload scripts.

0lav, 2023-11-26:
> That's unrelated to the docker issue. The curl libraries got an update a little while back, and is recommending that the plugins switch over to some new functions because the methods currently used [...]

I just open the docker console and do apt-get update then apt-get install curl to fix.

robotastic, 2024-12-14:
Closing - looks like Curl is back in the image

closed 2024-12-14 by robotastic