TrunkRecorder/trunk-recorder #858: Broadcastify System ID string/number config.json issue

james-coder, 2023-11-12:
I've recently compiled trunk-recorder on a WSL2 Ubuntu 20.04 system. (Using usbipd.)

I used a previously working config.json from another computer where `"broadcastifySystemId": "86753"` (Notice the quotes around the number.) This was working fine for me with no errors.

When I compiled it for the WSL2 Ubuntu 20.04 system, I get the error:
```
...
-------------------------------------
SYSTEMS
-------------------------------------
...
[2023-11-11 21:12:58.423169] (info)   Broadcastify API Key: BROADCASTIFYAPIKEY
[2023-11-11 21:12:58.423216] (error)   Failed parsing Config: [json.exception.type_error.302] type must be number, but is string
```
Which leads me to think it wants a int there instead instead of a string. I change it to be `"broadcastifySystemId": 86753` , but then I get this error:
```
...
-------------------------------------
PLUGINS
-------------------------------------
...
[2023-11-11 21:18:04.382722] (info)   Setting up plugin -  Name: broadcastify_uploader	 Library file: libbroadcastify_uploader.so
[2023-11-11 21:18:04.383107] (info)   Broadcastify Server: https://api.broadcastify.com/call-upload
[2023-11-11 21:18:04.383222] (error)   Failed parsing Config: [json.exception.type_error.302] type must be string, but is number
```

So basically, I getting one error message that says it should be an integer (in SYSTEMS), I change it to an integer, then I get another error message saying it should be a string (in PLUGINS).

The documentation at https://github.com/robotastic/trunk-recorder/blob/master/docs/CONFIGURE.md#system-object says:

**broadcastifySystemId**: if broadcastifyCallsServer is set System ID for Broadcastify Calls (**this is an integer**, and different from the RadioReference system ID)

The broadcastify plugins code is in https://github.com/robotastic/trunk-recorder/blob/47460b1582ec5187ed99eac0df93ae41ab93017f/plugins/broadcastify_uploader/broadcastify_uploader.cc

james-coder, 2023-11-12:
Here is my config.json:
```
{
    "ver": 2,
    "defaultMode": "analog",
    "sources": [
        {
            "center": 123456789,
            "rate": 20000000,
            "error": 0,
            "gain": 0,
            "ifGain": 24,
            "bbGain": 28,
            "digitalRecorders": 3,
            "analogRecorders": 8,
            "debugRecorders": 0,
            "driver": "osmosdr",
	    "logFile": true,
	    "consoleLog": false
        }
    ],
    "systems": [{
        "control_channels": [123456789, 123456789, 123456789, 123456789],
        "type": "smartnet",
	"maxDuration": 7200,
	"audioArchive": false,
	"recordUnknown": true,
        "talkgroupsFile": "tg.csv",
        "modulation": "fsk4",
	"squelch": -55,
        "shortName": "tgshort",
	"apiKey": "OPENMHZAPIKEY",
        "broadcastifyApiKey": "BROADCASTIFYAPIKEY",
	"broadcastifySystemId": 86753
    }],
    "uploadServer": "https://api.openmhz.com",
    "broadcastifyCallsServer": "https://api.broadcastify.com/call-upload"
}
```

Here is my output when running trunk-recorder:
```
$ ../trunk-recorder --config=config.json
[2023-11-11 21:18:03.187153] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------
[2023-11-11 21:18:03.187196] (info)   

-------------------------------------
INSTANCE
-------------------------------------
[2023-11-11 21:18:03.187204] (info)   Using Config file: config.json
[2023-11-11 21:18:03.187207] (info)   Trunk-Recorder: Version: 4.7.0
[2023-11-11 21:18:03.187209] (info)   Log to File: false
[2023-11-11 21:18:03.187213] (info)   Log Directory: logs
[2023-11-11 21:18:03.187220] (info)   Temporary Transmission Directory: /dev/shm
[2023-11-11 21:18:03.187223] (info)   Capture Directory: ./trunk-build/trunk-config
[2023-11-11 21:18:03.187226] (info)   Upload Server: https://api.openmhz.com
[2023-11-11 21:18:03.187228] (info)   Broadcastify Calls Server: https://api.broadcastify.com/call-upload
[2023-11-11 21:18:03.187231] (info)   Status Server: 
[2023-11-11 21:18:03.187233] (info)   Instance Key: 
[2023-11-11 21:18:03.187235] (info)   Instance Id: 
[2023-11-11 21:18:03.187237] (info)   Broadcast Signals: false
[2023-11-11 21:18:03.187240] (info)   Default Mode: analog
[2023-11-11 21:18:03.187242] (info)   Call Timeout (seconds): 3
[2023-11-11 21:18:03.187245] (info)   Control channel warning rate: 10
[2023-11-11 21:18:03.187247] (info)   Control channel retune limit: 0
[2023-11-11 21:18:03.187286] (info)   Phase 1 Software Vocoder: false
[2023-11-11 21:18:03.187308] (info)   Enable Audio Streaming: false
[... 127 lines cut ...]
[2023-11-11 21:18:04.362693] (info)   	 Decoder sink created!
[2023-11-11 21:18:04.366191] (info)   	 Analog Recorder Taps - initial: 8888 channel: 8888 ARB: 8888 Total: 8888
[2023-11-11 21:18:04.366820] (info)   	 Creating decoder sink...
[2023-11-11 21:18:04.367193] (info)   	 Decoder sink created!
[2023-11-11 21:18:04.370951] (info)   	 Analog Recorder Taps - initial: 8888 channel: 8888 ARB: 8888 Total: 8888
[2023-11-11 21:18:04.371634] (info)   	 Creating decoder sink...
[2023-11-11 21:18:04.371999] (info)   	 Decoder sink created!
[2023-11-11 21:18:04.375893] (info)   	 Analog Recorder Taps - initial: 8888 channel: 8888 ARB: 8888 Total: 8888
[2023-11-11 21:18:04.376579] (info)   	 Creating decoder sink...
[2023-11-11 21:18:04.376961] (info)   	 Decoder sink created!
[2023-11-11 21:18:04.377158] (info)   
-------------------------------------

[2023-11-11 21:18:04.377182] (info)   

-------------------------------------
PLUGINS
-------------------------------------
[2023-11-11 21:18:04.377192] (info)   Setting up plugin -  Name: openmhz_uploader	 Library file: libopenmhz_uploader.so
[2023-11-11 21:18:04.382609] (info)   OpenMHz Server: https://api.openmhz.com
[2023-11-11 21:18:04.382676] (info)   Uploading calls for: tgshort
[2023-11-11 21:18:04.382722] (info)   Setting up plugin -  Name: broadcastify_uploader	 Library file: libbroadcastify_uploader.so
[2023-11-11 21:18:04.383107] (info)   Broadcastify Server: https://api.broadcastify.com/call-upload
[2023-11-11 21:18:04.383222] (error)   Failed parsing Config: [json.exception.type_error.302] type must be string, but is number
```

e850205, 2023-11-12:
This was not fixed by #859  for me

james-coder, 2023-11-12:
@e850205 if you are getting an error, please share.

closed 2023-11-13 by james-coder