TrunkRecorder/trunk-recorder #868: Config minTransmissionDuration incorrectly parsed.

tadscottsmith, 2023-11-14:
Not sure if this applies to other numbers in the config.json, but it appears the new config parser is not correctly reading values for decimal numbers set in minTransmissionDuration.

`"minTransmissionDuration": 0.35,`
`[2023-11-14 13:03:16.468952] (info)   Minimum Transmission Duration (in seconds): 0`

taclane, 2023-11-14:
The parser seems to be operating as expected.  Sorta.  It appears do be doing some implicit conversion based on the default value, or where the value is being stored.  For instance:

`"minTransmissionDuration": 1.75,`
__
```
system->set_min_tx_duration(element.value("minTransmissionDuration", 0));
```
> `Minimum Transmission Duration (in seconds): 1`

vs. 

```
system->set_min_tx_duration(element.value("minTransmissionDuration", 0.0));
```
> `Minimum Transmission Duration (in seconds): 1.75`

I think the other way to avoid the conversion might be something like:
```
double tx_duration = element.value("minTransmissionDuration", 0);
system->set_min_tx_duration(tx_duration);
```

taclane, 2023-11-14:
Quickly scanning through the new config loader, other things that look like they may be affected by an unintentional conversion to integers include:
> PPM error, device gain settings, min/max call duration, squelch, and call timeout

tadscottsmith, 2023-11-28:
This appears to be resolved by 06b911c.

closed 2023-11-28 by tadscottsmith