TrunkRecorder/trunk-recorder #881: RPI setup not capturing or creating dir for audio to shortName directory

jeffbarclay, 2023-12-01:
Have RPI setup and capturing talk groups but not saving or creating the directory (shortName) - I do see wav files temporally created in a linix temp folder 

captureDir is set but never gets created or populated?

`
oncluding Recorded Call - Last Update: 8s	Recorder last write:3.93193	Call Elapsed: 9
[2023-12-01 10:56:54.006074] (info)   [montgomery]	4C	TG:          MCFR 7A1 Ops 1 (      4000)	Freq: 853.387500 MHz	Stopping P25 Recorder Num [1]	TDMA: false	Slot: 0	Hz Error: 53
[2023-12-01 10:56:54.008466] (info)   [montgomery]	4C	TG:          MCFR 7A1 Ops 1 (      4000)	Freq: 853.387500 MHz	- Transmission src: 63 pos:  0.00 length:  1.08 errors: 424 spikes: 20
O[2023-12-01 10:56:54.019488] (error)   [montgomery] Retuning to Control Channel: 853.937500 MHz
[2023-12-01 10:56:54.020027] (info)   	 - System Source 0 - Min Freq: 848.550000 MHz Max Freq: 858.450000 MHz
[2023-12-01 10:56:54.020776] (error)   [montgomery]	 Control Channel Message Decode Rate: 1/sec, count:  3
sox FAIL formats: can't open output file `/home/pi/trunk-build/montgomery/2023/12/1/4000-1701446205_853387500.0-call_4.wav': No such file or directory
[2023-12-01 10:56:54.036918] (info)   Combining: /dev/shm/montgomery/4000-1701446205_853387500.wav  into: ~/trunk-build/montgomery/2023/12/1/4000-1701446205_853387500.0-call_4.wav
[2023-12-01 10:56:54.037067] (info)   sox /dev/shm/montgomery/4000-1701446205_853387500.wav  ~/trunk-build/montgomery/2023/12/1/4000-1701446205_853387500.0-call_4.wav 
[2023-12-01 10:56:54.037112] (error)   Failed to combine recordings, see above error. Make sure you have sox and fdkaac installed.
Osox FAIL formats: can't open input file `/home/pi/trunk-build/montgomery/2023/12/1/4000-1701446205_853387500.0-call_4.wav': No such file or directory
OERROR: unsupported input file
[2023-12-01 10:56:54.072590] (error)   Failed to convert call recording, see above error. Make sure you have sox and fdkaac installed.
OOOOOOOOOO[2023-12-01 10:56:54.421035] (error)   P25 Parse error, s:  s0: 0 s1: 182 shift: 0 nac: 182 type: 65535 Len: 0
OOOOOOOOOOOOOOOOOOOO
`

jeffbarclay, 2023-12-01:
config.json:
 `                              
{
    "ver": 2,
    "captureDir":"~/trunk-build",
    "sources": [{
        "center": 853500000,
        "rate": 10000000,
        "ifGain": 7,
                "mixGain": 7,
                "lnaGain": 7,
                "digitalRecorders": 6,
        "driver": "osmosdr",
        "device": "airspy=0x26A464DC284F2C93"
    }],
    "systems": [{
        "control_channels": [853687500, 853862500, 853887500, 853937500],
        "type": "p25",
        "talkgroupsFile": "xmocotg.csv",
        "recordUnknown":false,
        "audioArchive": false,
        "transmissionArchive": false,
                "callLog":true,
        "talkgroupDisplayFormat":"tag_id",
        "shortName": "montgomery",
        "modulation": "qpsk",
        "controlWarnRate":0
    }],
    "plugins":[{
          "name": "rdioscanner_uploader",
          "library": "librdioscanner_uploader.so",
          "server": "http://localhost:3000",
          "systems": [{
                  "shortName": "montgomery",
                  "apiKey": "09bd3dcd-80af-4599-a5c7-42bfb1552bc4",
                  "systemId": 411
          }]
    }]
    
}
`

jeffbarclay, 2023-12-01:
hard path doesn't change anything and ownership ... its not creating the directory within trunk-build dir, even if I create it myself it doesn't write the wav files to it?

taclane, 2023-12-01:
6 airspy recorders at 10 MHz may also be a little intense for a raspi. Those `OOOOOOOOOOOOOOOOOOOO`'s leaking through at the end of the logs are indicating that it is receiving data faster than it can process it.  

Lower the sample rate if you can, but as long as you keep seeing that in logs, it's likely to cause issues down the line.

jeffbarclay, 2023-12-01:
after reducing recorders to 3 and sample rate to 2.5M its much cleaner but still no audio wav files in 'montgomery' folder

rosecitytransit, 2024-01-01:
If you want and are still experiencing the issue, I could possibly look into it

closed 2024-12-14 by robotastic