TrunkRecorder/trunk-recorder #885: Can Install an Older Version and If So, How?

Dewey3, 2023-12-04:
Can I do a "fresh" install of an older version of TrunkRecorder and if so, how?  (And will it still work if newer items like GNU Radio get installed?).  I really like the 9-15-2022, 10a2763 release since that is the last version that can keep up with rapidly changing user IDs on very busy systems.  I have an imaged fresh install of that version from back in late September 2022 that I can blow out, but I just want to create another brand new "fresh" install of that version on Bookworm and make some little system tweaks and changes.

tadscottsmith, 2023-12-04:
If you're manually building, you should be able to just switch to that commit before building. Something like:
```
git clone https://github.com/robotastic/trunk-recorder.git
cd trunk-recorder
git checkout 10a2763
```
That being said, it's probably a good reminder to try and dig in and fix the root cause...

Dewey3, 2023-12-06:
Thanks @tadscottsmith.  I got a bunch of `<command-line>: warning: "SPDLOG_FMT_EXTERNAL" redefined` type errors, but it is the version I wanted and is working.  Thanks again.

closed 2023-12-06 by Dewey3

rosecitytransit, 2023-12-30:
Alternatively, one can go to the [commit history](https://github.com/robotastic/trunk-recorder/commits/master), then "< >" button on right side for the desired commit, then green "Code" button then "Download ZIP"

Dewey3, 2024-01-16:
I know this topic is closed but for those who still see it ... during the 1/13-1/15 holiday weekend, I attempted to install the older version of TrunkRecorder using the `git checkout` line several times, all without success.  Every time I tried, the installation aborted around 13%, so my guess is that there have now been enough changes and upgrade that this older version will no longer work.  It installed fine each time I used the current version.  However, there is one difference.  This weekend I was trying to install it on Linux (Kali), but all of the other old version installs was to a RPi.  If I can break some spare time this week, I may try a RPi `git checkout` install just to see if I get the same failure.